subsurface/subsurface #1489: Import error from Cressi Newton computer

depaoli, 2018-07-11:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

After upgrading to 4.8.0 (MacOS DMG) I receive an error when I try to download the data from my Cressi Newton.
Rolling-back to 4.7.8 solved the issue.

### Operating system:

MacOS 10.11.6

### Subsurface version:
What version of Subsurface are you running?  
4.8.0
Does the same happen on another Subsurface version?  
No
Are you using official release, test build, or compiled yourself?  
Official DMG

### Steps to reproduce:

1) Press command-D to import from computer
2) Select Cressi/Newton computer, mount point and click "Download"

### Current behavior:

Error triggered

### Expected behavior:

Download succeed

### Additional information:

### Mentions:

mfbernardes, 2018-07-11:
@matteo we will need more info.

Can you run it from cmd line with -vv and on the download dialog, select the save logfiles options?

Then post the logfile and cmd output (remove you email from there first if you are using cloud).

depaoli, 2019-03-07:
I'm sorry but due to this problem and annoying serial driver management in MacOS (not related to Subsurface), I've opted for a VM running Linux to download the data from the Cressi and upload to the Cloud, and then using the MacOS client for the regular interactions.

dirkhh, 2019-03-07:
So the issue was from 8 months ago, you never added the information we requested... 
closing this as "lacking user input"

closed 2019-03-07 by dirkhh