subsurface/subsurface #1493: Qt 5.6 calling "subsurface/scripts/build.sh -desktop" fails

janiversen, 2018-07-11:
### Describe the issue:

- [ x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Just downloaded Qt 5.6 to test the travis problem.

Trying to generate ssrf, but get a build error:

[ 35%] Building CXX object core/CMakeFiles/subsurface_corelib.dir/qt-ble.cpp.o
/Users/jani/develop/src/subsurface/core/qt-ble.cpp:294:59: error: no member named 'createCentral' in 'QLowEnergyController'
        QLowEnergyController *controller = QLowEnergyController::createCentral(remoteDevice);

Seems build.sh forget to blend out BLE.

### Operating system:

macOS 10.13
Qt 5.6

### Subsurface version:

master

### Steps to reproduce:

1) clean dev. environment (rm -rf build)
2) run ./subsubsurface/scripts/build.sh -desktop

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

neolit123, 2018-07-11:
i think you can build with BLE and it would work for Qt 5.6.
http://doc.qt.io/qt-5/qlowenergycontroller.html#createCentral
^ `createCenter()` was added in Qt 5.7.

dirkhh, 2018-07-11:
This is interesting... see https://travis-ci.org/Subsurface-divelog/subsurface/builds/402866332
Which seems to indicate that current master builds against 5.5, 5.6, 5.7, and 5.9 (I couldn't find a distro that ships with 5.8...)

neolit123, 2018-07-11:
i don't see `qt-ble.cpp` being build in qt55 job, but i do see it in the qt56 job and there is no error.

ahh, it's because of this in qt-ble.cpp:
```
#if defined(Q_OS_MACOS) || defined(Q_OS_IOS)
	QBluetoothDeviceInfo remoteDevice = getBtDeviceInfo(QString(devaddr));
	QLowEnergyController *controller = QLowEnergyController::createCentral(remoteDevice);
#else
```

so in a way if the above is really needed 5.6 is too old for Mac OS.

neolit123, 2018-07-11:
i guess we can do something like:
```
#if (defined(Q_OS_MACOS) || defined(Q_OS_IOS)) && QT_VERSION >= QT_VERSION_CHECK(5, 7, 0) 
	QBluetoothDeviceInfo remoteDevice = getBtDeviceInfo(QString(devaddr));
	QLowEnergyController *controller = QLowEnergyController::createCentral(remoteDevice);
#else
	// this is deprecated but given that we don't use Qt to scan for
	// devices on Android, we don't have QBluetoothDeviceInfo for the
	// paired devices and therefore cannot use the newer interfaces
	// that are preferred starting with Qt 5.7
	QBluetoothAddress remoteDeviceAddress(devaddr);
	QLowEnergyController *controller = new QLowEnergyController(remoteDeviceAddress);
#endif
```

janiversen, 2018-07-12:
@neolit123 I like your #if suggestion, because if we support build.sh should work without modifying variables etc.

I added a check like that in #1488 for exact the same reason.

janiversen, 2018-09-14:
Closing this issue, as it is no longer a problem for me

closed 2018-09-14 by janiversen