subsurface/subsurface #1499: Cannot connect on first launch after update on Windows

tormento, 2018-07-12:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
I use nightly builds. After every manual update, i.e. unpacking the .exe and replacing older files, on the first launch I get the "automatic check for updates" windows. I can accept or decline but the following effect is that the first connection to cloud goes timeout and I get lower window red message "Cannot connect to cloud server, working with local copy". I have to close Subsurface and on the next use everything goes ok.

### Operating system:
Windows 10 x64

### Subsurface version:
Windows

### Steps to reproduce:
Have a functioning manually installation of Subsurface. Download updated build, uncompress it (7-ZIP is ok) and replace updated files. Launch and accept online update control.

### Current behavior:
Can't connect to cloud after accepting online update control.

### Expected behavior:
No problems connecting to cloud.

### Additional information:
Subsurface v4.8.0-80-gdc2f0f8e48e1,
built with libdivecomputer v0.7.0-devel-Subsurface-NG (02560a7e7fe82919d584d3edbf3876f90382052c)
built with Qt Version 5.9.2, runtime from Qt Version 5.9.2
built with libgit2 0.26.0
[...]
sync with remote https://cloud.subsurface-divelog.org//git/abc@abc.com[abc@abc.com]
git storage: Sync with cloud storage
delete proxy setting
Cloud storage: checking connection to cloud server
Checking cloud connection...
setGeometry: Unable to set geometry 120x30+885+485 on QWidgetWindow/'QMessageBoxClassWindow'. Resulting geometry:  450x96+885+485 (frame: 8, 31, 8, 8, custom margin: 0, 0, 0, 0, minimum size: 450x96, maximum size: 450x96).
git storage: Waiting for cloud connection (1 second(s) passed)
git storage: Waiting for cloud connection (2 second(s) passed)
git storage: Waiting for cloud connection (3 second(s) passed)
git storage: Waiting for cloud connection (4 second(s) passed)
git storage: Cloud connection failed
connection test to cloud server failed QNetworkReply::NetworkError(ProtocolInvalidOperationError) "Error transferring https://cloud.subsurface-divelog.org//make-latte?number-of-shots=3 - server replied: I'm a teapot" 418 "Linus does not like non-fat milk"
Cloud storage: unable to connect to cloud server
git storage: Can't reach cloud server, working with local data
git storage: Load dives from local cache
git load handling file 00-Subsurface

### Mentions:
@dirkhh

neolit123, 2018-07-12:
@tormento 

i've tried reproducing the steps with the latest master and with the continuous build binaries and i cannot get the first cloud connection to fail on Windows 7.
do you have a firewall or a proxy installed; i'd blame that on a networking issue.

mfbernardes, 2018-07-12:
@tormento 
I'm not much of a windows user, so I might be missing something, but the way you mention you are updating seems strange to me.

What "uncompress it (7-ZIP is ok) and replace updated files" really mean? 
Is there a reason not to simply use the installer?

tormento, 2018-07-12:
Why not? :) The issue is indipendent from the way you “install” the program
and, if not, there is something wrong in the installer or something wrong
in how the updated program reacts at first launch.

I think we are going away from the issue, calling firewalls and
installations when the real issue is timeout on cloud sync

neolit123, 2018-07-12:
@mfbernardes 

> What "uncompress it (7-ZIP is ok) and replace updated files" really mean?
Is there a reason not to simply use the installer?

this is fine on Windows in the case of Subsurface. one can just copy-paste our binaries around and it would just work.

@tormento

> The issue is indipendent from the way you “install” the program
and, if not, there is something wrong in the installer or something wrong
in how the updated program reacts at first launch.

the way you are doing it, should just work.

> I think we are going away from the issue, calling firewalls and
installations when the real issue is timeout on cloud sync

- did you try disabling your firewall completely? 
- are you connecting via a proxy?
- what's your connectivity to https://cloud.subsurface-divelog.org/
(try logging in there)

the timeout on cloud sync seems to be specific to your network setup. as i cannot reproduce the problem.

tormento, 2018-07-12:
I have no firewall or proxy between my computer and internet. Connection is
pretty fast too 100/20 with 5ms of ping on first esternal hop. As I keep on
saying, the error appears ONLY after the update check prompt and never more
until I update program again. That’s why I think there should be some
hidden big there. Perhaps the connection to cloud wants to start while the
prompt is still acting and it simply goes timeout or it triggers some
hidden bug.

neolit123, 2018-07-12:
assuming you update to a new version and then when the "automatic check for updates" window appears and you click "accept", do you try to File->"Open cloud storage" directly and aren't getting a red error message saying that you have to re-enter your credentials in preferences first?

tormento, 2018-07-12:
Let me try. Usually I quit and relaunch.

tormento, 2018-07-13:
Tried. Can't manually connect even.

I need to quit and relaunch.

tormento, 2018-07-13:
P.S: on subsurface-4.8.0-100-gfe38a768f2b1 I have error in map window:

MapWidget.qml failed to load!
The QML modules QtPositioning and QtLocation could be missing!

But they are there on the usual position as always.

janiversen, 2018-07-13:
#1501 should correct your mapWidget problem.

neolit123, 2018-07-13:
@tormento 

> Can't manually connect even.

this suggests a generic connectivity issue on your end.

try this in a browser:
https://cloud.subsurface-divelog.org/

it makes it difficult for me to help further.

tormento, 2018-07-13:
> @tormento <https://github.com/tormento>
>
> Can't manually connect even.
>
> ^ what do you mean with this quoted statement.

That on first launch after every update, even trying to manually open cloud
doesn’t resolve problem.

If it would be a connectivity issue, it would be always present or randomly
happening and not such repeatable.

neolit123, 2018-07-13:
you did not answer this question from me:
> assuming you update to a new version and then when the "automatic check for updates" window appears and you click "accept", do you try to File->"Open cloud storage" directly and aren't you getting a red error message saying that you have to re-enter your credentials in preferences first?

do you getting a red warning?

tormento, 2018-07-13:
No need to re-enter credentials. And yes, the error is red.

neolit123, 2018-07-13:
> No need to re-enter credentials. And yes, the error is red.

in my case i always have to re-enter credentials.

you can try this:

- close subsurface
- open the start menu and enter "regedit" to start it
- in regedit open HKEY_CURRENT_USER / Software
- rename the Subsurface folder to Subsurface_ (this will backup your settings)
- start subsurface

see if the error happens again.

NOTE: you can always rename your old settings Subsurface_ in regedit back to Subsurface, but make sure any new Subsurface folders are deleted in there first.

neolit123, 2018-07-13:
your message did not reach the github thread.

> I don't use installer, thus the registry is not touched after every update. It seems very strange that you have to reenter credentials every time too.

that's not true.
each update writes a new version in the registry.

> I keep on being convinced that the update check window is the culprit of the problem. If some of the developers could disable it for nightly, we would have a proof or not of what I am saying.

could be, but i don't think so.
/cc @dirkhh

cyberdiamond, 2018-07-23:
I confirm that I see the same issue even using the full installer.

mikeller, 2026-10-17:
Hi @tormento,

This matches a recurring pattern we've seen with concurrent startup dialogs (consent/update modals) interfering with connection timing at launch, but it's a timing-dependent race we've never been able to reliably reproduce or pin down, and this report is now 8 years old with no further detail since. We're closing this as stale rather than leaving it open indefinitely.

Please reopen with a fresh, detailed repro (ideally a verbose log) if you're still seeing a first-launch connection failure after updating on a current build.

closed 2026-10-17 by mikeller