subsurface/subsurface #1500: Map not working under Windows 10 with 4.8.0-100

sfuchs79, 2018-07-12:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

With latest continuous build 4.8.0-100 and with my own MXE based Windows build based on master the map is not working.
I only see the (well known) error message in the map window.
I guess this comes from one of the latest commits because my last build with working map can't be longer than 2-3 days ago.

### Operating system:

Windows 10

### Subsurface version:

4.8.0-100

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

@dirkhh @janiversen @jbygdell @bstoeger

janiversen, 2018-07-12:
Can you please mail me the log, that might reveal what the problem is. Thanks in advance.

janiversen, 2018-07-12:
your own windows build, might not have catched up with the CMake changed. There is a new file
subsurface/subsurface-helper.cpp that replaces subsurface-*-helper.cpp

sfuchs79, 2018-07-12:
Error message from 4.8.0-100 from Travis:
qrc:/qml/MapWidget.qml:5:1: module "org.subsurfacedivelog.mobile" is not installed
QUrl("qrc:/qml/MapWidget.qml") failed to load with status: QQuickWidget::Status(Error)

janiversen, 2018-07-12:
That looks exactly as if your build does not use the new setup (files in subsurface)

The registrations was moved to subsurface/subsurface-helper.cpp

Let me just do another test, now my patch is merged (I tested on my local branch).

janiversen, 2018-07-12:
just rebuild master (to be sure), it works fine on a Mac.

sfuchs79, 2018-07-12:
Have you tested the Windows build from Travis?
The error from above is not from "my" build but from using the Travis CI Windows installer. This is what I installed to exclude anything coming from my build environment.

janiversen, 2018-07-12:
No I do not have access to windows, I am a Mac and iOS type of developer.

But based on your information, it looks like travis did not pickup the merge correctly, there are no changes in the actual code, just moved all qmlRegisterCode to register_qml_type().

@dirkhh if this is not something easy, then lets wait with this merge until after the release. It really looks like the "only source move" merge.

janiversen, 2018-07-12:
watt = revert

sfuchs79, 2018-07-12:
Would be great if s.o. else could also test the Windows CI installer.

Unfortunately I won't be able to bisect or simple revert some of the latest commits and test again. Started a rebuild of my MXE plus will got to bed now :-)

janiversen, 2018-07-12:
@dirkhh did a little analysis if the windows did not get the changes subsurface directoty but in subsurface/* directories this oroblem would happen

neolit123, 2018-07-12:
@sfuchs79 i was testing the `-100` Win32 build from our CI one hour ago and the map did no throw errors for me.

could it be a case that your MXE build and the `-100` build somehow mixed?

> qrc:/qml/MapWidget.qml:5:1: module "org.subsurfacedivelog.mobile" is not installed
QUrl("qrc:/qml/MapWidget.qml") failed to load with status: QQuickWidget::Status(Error)

the above message is very weird because i think for that to happen the file `mapwidget.cpp` would have to be missing from the build.

neolit123, 2018-07-12:
> could it be a case that your MXE build and the -100 build somehow mixed?

oh, wait a minute. i installed the `-100` over my previous install.
the map is indeed broken in `-100` installer.

janiversen, 2018-07-12:
actually it is more likely that qmlRegisterType which used to be in mapwidget is not called. that call moved to subsurface_helper.cpp

neolit123, 2018-07-12:
submitted fix.

PR #1501 (2018-07-13): map-widget: fix chronology of QML class registration

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

The QML types needs to be registered before the MainWindow
instance is created on the desktop version, otherwise
the MapWidget instance will be created and it will fail
with a missing QML namespace.

In subsurface-helper.cpp, move register_qml_types() from run_ui()
to init_ui(), as later in init_ui(), MainWindow is insta [... cut]
closed 2018-07-13 by neolit123