subsurface/subsurface #1514: Dive media files: Add "Open file location in file manager" to context menue

sfuchs79, 2018-07-15:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [x] New feature request
- [ ] Discussion request

### Issue long description:

This is a small feature request for the dive media files implementation which I would like to discuss and maybe also ask s.o. to implement it.
It's simply that I would like to have a context menu entry to open the selected media files folder in the systems default file manager. Use case is: I found a nice picture or video of a dive which I want to share e.g. by copying to my Dropbox of sending it by email. That would work much faster if I immediately can open the correct folder with the file(s).

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

@bstoeger @atdotde

bstoeger, 2018-07-18:
Doesn't look too hard.

What should happen if multiple files belonging to different folders are selected? Open all directories?

Generally, I feel that the context-menu should be moved down from `TabDivePhotos` to `DivePictureWidget`. From the latter we have easy access to the item that is below the mouse cursor and the user could simply right-click > open file in filemanager, without need for explicit selection.

atdotde, 2018-07-18:
You can open the directory via QDesktopServices::openUrl( QUrl::fromLocalFile( path ) );

If you want to highlight files, this might be relevant: https://stackoverflow.com/questions/3490336/how-to-reveal-in-finder-or-show-in-explorer-with-qt

commit 4f42e4fd (2018-07-20): Dive media: implement "Open folder of selected media files"

Add a context-menu entry to TabDivePhotos which opens the folder(s)
of all selected files.

Fixes #1514.

Suggested-by: Stefan Fuchs <sfuchs@gmx.de>
closed 2018-07-20 by atdotde