subsurface/subsurface #1546: Improve exception safety with smart pointers

elfring, 2018-07-30:
Would you like to [wrap any pointer data members](https://github.com/Subsurface-divelog/subsurface/blob/d3af836467a2c1a4913eb9e444632d834c7a335c/desktop-widgets/templateedit.cpp#L11 "Update candidate: TemplateEdit class") with the class template “[std::unique_ptr](https://en.wikipedia.org/wiki/Smart_pointer#unique_ptr "Description for the usage of smart pointers")”?

tcanabrava, 2018-07-30:
Nope.

tcanabrava, 2018-07-30:
I think this deservers a bit more of explanation.
This code is not completely written in C++, most of it is in fact C, and I cannot use std::unique_ptr in c, so we need to receive raw pointers in the c++ code. We also don't want to englobe qt pointer's in unique_ptr because qt manages the pointers itself. so using a unique_ptr in every pointer data member will only create bugs and double frees.

closed 2018-07-30 by tcanabrava

elfring, 2018-07-30:
I suggest to [reconsider the resource management in affected source files](https://isocpp.github.io/CppCoreGuidelines/CppCoreGuidelines#r11-avoid-calling-new-and-delete-explicitly "C++ core guideline “R.11: Avoid calling new and delete explicitly”") once more.

elfring, 2018-07-30:
>The c++ core guidelines don't work at all if you are interfacing with C.

I find this kind of feedback (partly) inappropriate.

* I propose to take another look at affected details together with corresponding development challenges.
* I tried to point out that [the class “TemplateEdit” is an update candidate](https://github.com/Subsurface-divelog/subsurface/blob/d3af836467a2c1a4913eb9e444632d834c7a335c/desktop-widgets/templateedit.cpp#L11 "Reconsider implementation of TemplateEdit class") (because its destructor contains two delete statements).

:crystal_ball: Further implementation areas can be improved after related software clarifications.

dirkhh, 2018-07-30:
To be honest, I have no idea where you are coming from, I have no idea why you are engaging with the project. Are you trying to address a bug? Are you on a developer trying to rid the world of bad pointers? This looks a lot like a drive-by-shooting, so I'm with @tcanabrava here...

elfring, 2018-07-30:
>To be honest, I have no idea where you are coming from,

It could have happened that you noticed some of my contributions in other areas.

>I have no idea why you are engaging with the project.

I am trying to improve various software based on results from source code analysis.

>Are you trying to address a bug?

I propose to fix a known programming mistake also for this software.

>Are you on a developer trying to rid the world of bad pointers?

:crystal_ball: It seems so.

dirkhh, 2018-07-30:
Thanks for that explanation @elfring 
We really appreciate your interest in our project.
I think we will politely decline your offer to tell us that we are wrong.
Thanks and all the best