subsurface/subsurface #1547: Completion of error handling

elfring, 2018-07-30:
Would you like to add more error handling for return values from functions like the following?
- [calloc](http://pubs.opengroup.org/onlinepubs/9699919799/functions/calloc.html "Memory allocation") ⇒ [ostctools_import](https://github.com/Subsurface-divelog/subsurface/blob/826c01d0a00253727942ae26e876d509a72908b3/core/ostctools.c#L34)
- [fclose](http://pubs.opengroup.org/onlinepubs/9699919799/functions/fclose.html "Close a stream") ⇒ [local_hw_ostc_device_eeprom_write](https://github.com/Subsurface-divelog/subsurface/blob/a312e53f0ce47dd2797070dff37e51fbfafd489a/core/configuredivecomputerthreads.cpp#L115)

neolit123, 2018-08-08:
i think the` fclose() `change is not really needed, but if you wan you can submit a PR for the `calloc()` checks. :+1: 
closing this issue as the tracker is usually for features and bugs.

you can also subscribe to the Subsurface developer mailing list and post suggestions there:

closed 2018-08-08 by neolit123

elfring, 2018-08-09:
The bug is incomplete exception handling, isn't it?

I suggest to avoid [ignorance of return values](https://cwe.mitre.org/data/definitions/252.html "Checking more return values") a bit more.
Would you like to detect every error situation as early as possible?

Do [known source code analysis approaches](https://en.wikipedia.org/wiki/List_of_tools_for_static_code_analysis#C.2C_C.2B.2B "Tools for code analysis") point any more open issues out also for this software?

elfring, 2018-08-09:
Can fixes evolve into specific source code formats?

* Are you interested to apply [aspect-oriented software development](https://aspectc.org/ "Software development with AspectC++?")?
* How do you think about to encapsulate error detection and corresponding exception handling as [a reusable aspect](https://en.wikipedia.org/wiki/Aspect_%28computer_programming%29 "Combination of a pointcut with advice") in this software?

dirkhh, 2018-08-09:
I have come to the conclusion that @elfring is a troll, let's not waste our time here

atdotde, 2018-08-09:
I was about to do the same.