subsurface/subsurface #1557: FB posting stopped working

bushibot, 2018-08-03:
I noticed with the 4.8.1 build that FB posting has stopped working. Login is passing correctly (though dialogs, method, and browser memory are all outdated).
After logging to FB via subsurface and trying to share/facebook and filling in form, no post message returns (previous behavior), no error message returns, but no FB post occurs.

janiversen, 2018-08-03:
Could you please post a log file. Do you use mobile or desktop ?

bushibot, 2018-08-03:
Desktop, windows 10. I don't think you can post to FB from mobile?
Log attached.  Two note in looking at the log readout, the album dives does exist, and the action was not user canceled :).
Looks like the relevant bit is:
[subsurface_out.log](https://github.com/Subsurface-divelog/subsurface/files/2258816/subsurface_out.log)
("http://www.facebook.com/connect/login_success.html#access_token=EAAGFAucfBAgBADjSW8Dhac8wVaWfkTZCdWA3KJgKxFkHp74QtB5lvBYrSIZC4yYRp3ZCAV25LHf19EKPaZBnf3OyyUr96NmZC9ltDfd4nByZA3hBmjDI0PRMiL4TPihIY0iZB34L5DZBBwbKYCPupYiJ9Vq5GyHNwW14AE1ZAfbMEWMhljF2kxNTk&expires_in=5183696&granted_scopes=public_profile&denied_scopes=")
Got securityToken "EAAGFAucfBAgBADjSW8Dhac8wVaWfkTZCdWA3KJgKxFkHp74QtB5lvBYrSIZC4yYRp3ZCAV25LHf19EKPaZBnf3OyyUr96NmZC9ltDfd4nByZA3hBmjDI0PRMiL4TPihIY0iZB34L5DZBBwbKYCPupYiJ9Vq5GyHNwW14AE1ZAfbMEWMhljF2kxNTk"
Requesting user id
User id requested successfully: "10205345011978904"
Current url call QUrl("http://www.facebook.com/connect/blank.html#_=_")
Response without access token!
Starting to upload the dive to facebook
Grabbing Dive Profile pixmap
Starting to request the album id QUrl("https://graph.facebook.com/me/albums?access_token=EAAGFAucfBAgBADjSW8Dhac8wVaWfkTZCdWA3KJgKxFkHp74QtB5lvBYrSIZC4yYRp3ZCAV25LHf19EKPaZBnf3OyyUr96NmZC9ltDfd4nByZA3hBmjDI0PRMiL4TPihIY0iZB34L5DZBBwbKYCPupYiJ9Vq5GyHNwW14AE1ZAfbMEWMhljF2kxNTk")
Reply for the album id
Album with name "dives" doesn't exists, creating it.
It was not possible to create the album with name "dives"
Reply was:  {
    "error": {
        "code": 10,
        "fbtrace_id": "BB4ET04+Wx5",
        "message": "(#10) Application does not have permission for this action",
        "type": "OAuthException"
    }
}

But we are still able to upload data. Album name will be 'Subsurface Photos'
Starting to upload the dive to album "dives" id ""
Upload finish
Dive upload failed "{\"error\":{\"message\":\"(#200) Requires extended permission: publish_actions\",\"type\":\"OAuthException\",\"code\":200,\"fbtrace_id\":\"Aj0VpVSFWqN\"}}"
Starting to upload the dive to facebook
User cancelled.
Starting to upload the dive to facebook
User cancelled.

janiversen, 2018-08-03:
Thanks for the log, I made some changes to the FB setup, but I am not sure they made it to 4.8.1. I will however have a look at your problem during the weekend.

neolit123, 2019-02-01:
we are  removing the FB feature:
https://github.com/Subsurface-divelog/subsurface/pull/1958

closed 2019-02-01 by neolit123

bushibot, 2019-02-01:
Removal? That sucks. I use the FB post all the time. I would use it more if I didn’t have to go through the login hoops every time. Social media sharing is kind of a relevant thing these days.... it should be easy and a standard.