subsurface/subsurface #1559: No package for Ubuntu 18.04 i386

bubjo, 2018-08-04:
At the moment it is not possible to install Subsurface under Ubuntu 18.04 32-bit.

bubjo, 2018-08-09:
That is the 64-bit version, but I need the 32-bit version.

dirkhh, 2018-08-09:
On Thu, Aug 09, 2018 at 09:18:37PM +0000, bubjo wrote:
> That is the 64-bit version, but I need the 32-bit version.

Build from source. I don't think there's another path right now.

/D

sarthakthakur24, 2018-08-13:
You can build from source as suggested by @dirkhh or how about you try this to use snaps instead.
https://snapcraft.io/subsurface

dirkhh, 2018-08-13:
I didn't realize that the Snap was available in 32bit as well. My fault. And thanks for pointing this out @sarthakthakur24

closed 2018-08-13 by dirkhh

bubjo, 2018-08-14:
Sorry, but then Snap-Version does not work under Bionic 32-bit.

mturkia, 2018-08-18:
there is now a 32 bit version on launchpad for bionic. Would be great to hear if everything works on it.