subsurface/subsurface #1564: Subsurface on Mac not working

diver1008, 2018-08-07:
Running OS 10.12.6
Tried TreiberOSX_v2_2_16 and
TreiberOSX_v2_2_16 19-00-47-917
Suunto Cobra 2
Cable from https://www.smartinterface.de
And Suunto original cable

Hi folks,

I have the following problem when I try to download dives (with any of the cables).  I use Subsurface normally but also tested with MacDive.  Both of them give me the same errors.  See pics below.  
 
So here some facts and stuff I tried:

Unit flashes two or three times green when I try to download.  Next the enclosed box appears (see pics below).

I installed FTDI.  Noted that on the smartinterface.de site there is a different version as on the http://www.ftdichip.com site.  Tried both. Rebooted 3 times after install no luck either. 

Was in some forums to see if I find help.  Found some stuff but had no results.
I run:
ls -l /dev/cu*  
get
crw-rw-rw-  1 root  wheel   20,   1  5 Aug 18:49 /dev/cu.Bluetooth-Incoming-Port
or if I run ls /dev/tty.* I get:
/dev/tty.Bluetooth-Incoming-Port	/dev/tty.usbserial-00002014
and with the Suunto cable I get:
/dev/tty.Bluetooth-Incoming-Port	/dev/tty.usbserial-ST04MTY9

I also found this:
“And one thing that I noticed when trying to reproduce this. It appears that you have to reboot your Mac after installing the driver. And I read two threads in which people complained that they had to reboot TWICE for it to work.   So if: kextstat | grep FTDI results in no output, please try to reboot, run the command again, and if it still doesn't show, try a second reboot...”

I get no response and rebooting 3 times > does not solve it.

In the System information I see several FTDI things under Installations (pic 3):
FTDI USB Serial Driver:
  Version:	
  Source:	3rd Party
  Install Date:	05/08/2018, 12:04

FTDI USB Serial Driver:
  Version:	
  Source:	3rd Party
  Install Date:	05/08/2018, 12:40

FTDI USB Serial Driver:
  Version:	
  Source:	3rd Party
  Install Date:	05/08/2018, 18:41

FTDIUSBSerialDriverInstaller:
  Version:	
  Source:	3rd Party
  Install Date:	05/08/2018, 12:42

FTDIUSBSerialDriverInstaller:
  Version:	
  Source:	3rd Party
  Install Date:	05/08/2018, 19:00

That is what I did/know so far, any help or suggestions?

Thanks. 

[image]
[image]

dirkhh, 2018-09-05:
Sorry, this bug report slipped through the cracks.
The latest test builds should help address this as they include the FTDI driver directly.
Instead of /dev/tty.... you use ftdi as the Connection text.
This only works with the test build, not with the last release (4.8.1) -> (we are planning a 4.8.2 release fairly soon).
If you want to try a test build, you can find them here:
https://github.com/Subsurface-divelog/subsurface/releases/tag/continuous

diver1008, 2018-09-14:
> <reply+02839678fec27cea56ef58c027733814c3036c9dde87bdc492cf0000000117a7fed79
> 2a169ce14c2f1fc@reply.github.com>
> (#1564)

Thanks, will try it out.

With kind regards,
 
Eric
 
--
:-))
IMPORTANT: This email is intended for the use of the individual addressee(s)
named above and may contain information that is confidential privileged or
unsuitable for overly sensitive persons with low self-esteem, no sense of
humour or irrational religious beliefs. If you are not the intended
recipient, any dissemination, distribution or copying of this email is not
authorized (either explicitly or implicitly) and constitutes an irritating
social faux pas. Unless the word absquatulation has been used in its correct
context somewhere other than in this warning, it does not have any legal or
grammatical use and may be ignored. No animals were harmed in the
transmission of this email, although the kelpie next door is living on
borrowed time, let me tell you.
 
 Those of you with an overwhelming fear of the unknown will be gratified to
learn that there is no hidden message revealed by reading this warning
backwards, so just ignore that Alert Notice from Microsoft. However, by
pouring a complete circle of salt around yourself and your computer you can
ensure that no harm befalls you and your pets. If you have received this
email in error, please add some nutmeg and egg whites, whisk and place in a
warm oven for 40 minutes.

jb2cool, 2020-03-08:
@dirkhh the original poster reported back that this was fixed as of 4.8.3 so this 'bug' can likely be closed now.

Looking through the list for any quick wins, even if it's just closing 'bugs' that are not longer needing to be looked at.

dirkhh, 2020-03-08:
Thanks for going through those old issues, @jb2cool - much appreciated and one of those tasks that I often don't find enough time for. This is a great way to help the project!

closed 2020-03-08 by dirkhh