subsurface/subsurface #1576: Insufficient quoting in HTML export

dirkhh, 2018-08-14:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

from the user forum:

Got the following two errors in the JS console:
```
list_lib.js:813 Uncaught ReferenceError: trips is not defined
    at getItems (list_lib.js:813)
    at window.onload (dives.html:109)
```
and 
```
dives.html_files/file.js:63 Uncaught SyntaxError: Unexpected identifier
```
Looking at line 63, I see I have some double quotes inside a string, which apparently weren't sanitized/escaped before including in file.js. Changing to single quotes and it's fine.
So, for those of us who don't know a) much Javascript, and b) what is and is not escaped upon being written to file.js, what else should I be careful of using in various free-form fields?

neolit123, 2018-08-15:
cannot reproduce the user issue.
everything is escaped correctly via `membuffer.c::put_quoted()`

jmedway, 2018-08-22:
One thing to make clear as the original reporter: The issue might be restricted to the "Trip Location" field not being sanitized. Any chance that is handled differently?

neolit123, 2018-08-23:
i will submit a fix in a minute.

PR #1596 (2018-08-23): save-html: make sure that some strings are quoted

### Describe the pull request:

- [x] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

- filenames for pictures
- trip locations
- event names (not really needed)

### Changes made:
- use `put_quoted()` and `write_attribute()` so that attribute values are quoted.

### Related issues:

Fixes #1576

### Additional information:

none

### Release note:

```
Desktop/Export: fix bug related to quoted text when exp [... cut]
closed 2018-08-23 by dirkhh