subsurface/subsurface #1581: Display divesites after using filter tool

willemferguson, 2018-08-16:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Running subsurface version 4.8.1-249 on Ubuntu 18.04
After filtering a dive list, only the dives fitting the key are listed in the dive list, as it should be.
However, the divesite map is not updated to agree with the (now shortened) dive list that fit the key of the filter. This actually appears to work on some versions of Subsurface, e.g. apparently in Windows, though I could not verify this.
When I open the attached divelog (testMap.xml) with 6 dives and filter it for "Drysuit", then only the most north-easterly location should show. But all four location remain on the map (see attached testMap.jpg).

### Operating system:

Ubuntu 18.04

Dunno

### Subsurface version:
 V4.8.1-249
 Yes, 4.8.1
 Self-compiled

### Steps to reproduce:

1) open attached dive log.
2) Go to Log->Filter divelist and select the two keys with "Drysuit"
    Only dives from one divesite should be visible in the dive list.
3) Inspect any changes in the map content.

### Current behavior:

As explained above
### Expected behavior:

The dive map should only show one divesite and not four.

### Additional information:

### Mentions:

@neolit123

[image]
[testMap.zip](https://github.com/Subsurface-divelog/subsurface/files/2294733/testMap.zip)

bstoeger, 2018-08-17:
I currently don't have a working map (c.f. #1409), but this shouldn't be too hard.

In `reloadMapLocations()` we could first collect all dive sites with non-filtered dives (`!dive->hidden_by_filter`).

Then, on filter change repopulate the map by calling `reloadMapLocations()`.

neolit123, 2018-08-17:
now i understand the issue.
this seems doable. i will look into it when i get a minute.

closed 2018-08-18 by bstoeger