subsurface/subsurface #1590: Setting location in mobile resulted in GPS data loss

emonty, 2018-08-19:
### Describe the issue:
- [ x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Adding location in mobile app after adding GPS coordinates results in loss of GPS coordinates.

### Operating system:
Android 9

### Subsurface version:
2.1.0(4.8.0.0)

### Steps to reproduce:
At the dive site, I synced dives from my dive computer over bluetooth. Then I edited the dives and used the "Use current GPS location" checkbox to add GPS information to the dive. I then saved the dive and syncced to the cloud.

Later, no longer at the site, I went to edit the dive and add a location name, still using the mobile app. When I added the Location name, the GPS data was blanked out- and there was not an option for "create location using current GPS coordinates" like in the desktop version.

I solved the specific issue by cloning the git repo and looking at the git diff to see what the GPS coordinates were before the commit that replaced them with a location that didn't have coordinates. (btw - I cannot possibly express enough gratitude that the data format is readable text in git repos)

I'd be happy to work on a fix, depending on what the right fix is.

### Current behavior:
the thing from above

### Expected behavior:
It would be GREAT to be able to add a new location and keep the GPS. If that's not possible, a validation warning about information loss would be better - I can always wait to add a location name until later.

### Additional information:

### Mentions:

mfbernardes, 2018-08-20:
The way it currently works is:
When you change the Location field it loads the GPS for that location. So if you enter a new location it will load empty GPS coordinates.

The effect is that the GPS field will always be replaced if you change the location name.

This has confused other people before as well, so we should make this more "user friendly"... but I'm not sure what is the right thing to do yet, maybe just a confirmation dialog like you mentioned.

ahvanderboon, 2018-12-20:
Hi, 
Although the GPS tracking is a wonderful feature, many locations can be found online. The mobile app however has difficulties saving coordinates pasted or entered manually, or they indeed disappear when entering or updating the dive site name.

Would suggest the dependency between the site name is not right, unless the site name in the database already had a GPS location. Also when the use current GPS location is not ticked, i would suggest to adopt the manual entered data.