subsurface/subsurface #1591: Add unit selection preference to mobile app

emonty, 2018-08-20:
### Describe the issue:
- [ ] Bug
- [ ] Change request
- [ x ] New feature request
- [ ] Discussion request

### Issue long description:
For reasons surpassing my personal understanding, my partner prefers Imperial units. And yet, the dive detail views in the mobile app only show the dive information in metric. Now, clearly metric is superior in virtually every way, but yet still I'd like to help those who have not yet transitioned their thinking.

### Operating system:
Android

### Subsurface version:
2.1.0(4.8.0.0)

### Steps to reproduce:
Open a dive in the mobile app
Try to see units in Imperial.

### Current behavior:
Only gloriously understandable metric units

### Expected behavior:
Confusing and nonsensical imperial units are displayed instead.

### Additional information:
I started digging through the code, but I'll have to admit I have some things to learn about QML+Android and could use some structural pointers.

### Mentions:

janiversen, 2018-08-20:
This is something that is in the works. buth the pointers are easy look at the directory mobile-widgets qmlmanager.cpp as well as the qml directory.

mfbernardes, 2018-08-20:
Hello @emonty 

The way it should currently works is:
* If you have synched with cloud, it will use your selections from the Desktop
* Otherwise it will be based on your locale.

We've seen some problems with certain locales, so.
- Have you synched with cloud both on Desktop and mobile?
- What is your locale? You can simply copy the log by going to About, and clicking "Copy logs to clipboard" (or equivalent in your language).

janiversen, 2018-08-20:
@mfbernardes 

are you sure about the sync bit? qPrefUnit only loads local settings, but I did not check if that is overwritten in all places where used.

mfbernardes, 2018-08-20:
@janiversen hmm... true, for some reason I had in mind it worked and didn't check the code now.

So, yeah, it will be Locale based only.

jbygdell, 2018-09-17:
Derek assign: me

PR #1705 (2018-09-22): Update divelist when changing units

### Describe the pull request:

- [X] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

We need to explicitly refresh the divelist when switching between
metric and imperial unit systems. Or the changes will not be visible until
we restart the app or scroll outside of what's in the current cache.
This will update both the divelist view and the dive profiles to show the new units.

### Changes made:

### Related issues:

Rel [... cut]
closed 2018-09-22 by dirkhh