subsurface/subsurface #1593: reading out OSTC4 gives crazy CNS values

antiker1, 2018-08-20:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Since some version of subsurface (but at least in the current version), when I read out an OSTC4 it gives crazy CNS values like 700% and more.
I just noted this by chance when I diffed the freshly read out XML with the one from last year (because it seems to happen quite often that subsurface mangles around with values it shouldn't touch).
It's the same dives than with an older version where it didn't show these crazy CNS values, so why should they be different?

Any idea how to fix that?

Thanks

### Operating system:
Linux
OSTC4

### Subsurface version:
4.8.0

glance-, 2018-08-20:
Dump out the raw dive from the dc with dctool. Something like `dctool ... download -f RAW -o buggy-dive-data-%n.dive`

antiker1, 2018-09-02:
Where can I get that from? It doesn't seem to be part of the subsurface packages?

janmulder, 2018-09-04:
Dctool is from Jef's website: `http://libdivecomputer.org/builds/stable/linux/`

mikeller, 2026-10-17:
Hi @antiker1,

We never got the raw dctool dump this would have needed to pin down exactly, so I can't point to one specific commit that fixes this report. What I can say: the CNS/OTU calculation has been revisited and fixed repeatedly since you reported this, a sign-error fix in PR #2227, a PSCR-specific fix in PR #3216, a CCR-dive fix in PR #4467, and further PO2-resolution hardening for CCR/PSCR in PR #5034. Given how much work has gone into that calculation since, it's likely this was addressed along the way, but I can't confirm that with certainty against your specific OSTC4 data.

Closing this on that basis, with real uncertainty attached. Please reopen with a fresh dctool dump if you (or anyone else) still see implausible CNS values reading out an OSTC4 on a current build, that would let us check directly rather than relying on this general history.

closed 2026-10-17 by mikeller