subsurface/subsurface #1598: OSX has an extra unused menu called "Show Tab Bar"

neolit123, 2018-08-25:
### Describe the issue:

- [ ] Bug
- [x] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

forum thread:
https://groups.google.com/d/msgid/subsurface-divelog/04475bbd-31bb-4def-9d64-c3eb9c28c631%40googlegroups.com?utm_medium=email&utm_source=footer

> Not sure if this is Mac or general, but I can't find what the functionality of the "Tab Bar" is supposed to be. 
I select 'Show Tab Bar" and I get the one item which is labeled with my XML file. Thinking, "ah, I can open another profile and ..." I open another, and it closes the first , so no tab functionality there. So, basically left puzzled, especially since I cannot get the "Show All Tabs" item to be not greyed out, so I've no way of seeing what else should appear.

we should probably attempt to disable this functionality as described here:
https://forum.qt.io/post/411727

### Operating system:

OSX

### Subsurface version:

latest?

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

@dirkhh @janiversen

janiversen, 2018-08-25:
Niw that you made me aware of the menu, it irritates me :-)

I am all for removing it.

janiversen, 2018-08-25:
Studying it a bit more, it turns out that the base problem (at least on my vm) is the info.plist (inContrnts)

the entry “CFBundleDevelopmentRegion” “English” causes the menu to appear. I am not sure why we specify CFBundleDevelopmentRegion (I have never used it), so I do not know which consequences it will have if we remove it.

dirkhh, 2018-08-25:
So as I'm sure you found out, that key sets the development language and it acts as a last fallback for defining what language and what resources to use.
I don't think we need it because of the way we handle language selection, but of course I wonder why this would have the side effect that you describe...

janiversen, 2018-08-25:
I wondered about the “why” too, but I found it in a google search, and tried it (still on my limited vm).

I still have a funny feeling that the fallback might be the cause of some of the nunber problems I have (en_ES), but it is just a feeling.

dirkhh, 2018-08-25:
Let's remove it and see what happens :-)

janiversen, 2018-08-26:
hmmm osx trixed, everything works different when you have an ES installation.

This need some more digging, and most likely some native calls (there is an objective C call that disables the menu entry, but calling that from C++ is anything but funny).

I am not able to work on this for a while, if anybody feels like picking it.

dirkhh, 2018-08-26:
I have played with calling ObjC interfaces before. It's actually not too bad.
What API needs to be called for this? And frankly, why isn't It doing that anyway?

dirkhh, 2018-09-05:
this hasn't moved anywhere. one of the things to still investigate