subsurface/subsurface #1607: Segfault when opening another logbook

p-neves, 2018-08-28:
### Describe the issue:
When I start Subsurface and open another logbook, the program segfaults.

- [X ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When I start Subsurface it opens my default logbook (as it should). When I try to open another logbook (File/Open logbook) the application segfaults.

### Operating system:

Arch Linux

### Subsurface version:

4.8.1

Not with appimage 4.7.8

Self compiled

4.8.1-303-g66aeaddd0f9c

### Steps to reproduce:

1) Start Subsurface
2) Click File | Open logbook
3) Navigate to folder and select file to open

### Current behavior:

Subsurface segfaults

### Expected behavior:

Open selected file

### Additional information:

gdb output:

(gdb) run
Starting program: /home/xxxxx/Diveapps/subsurface/build/subsurface 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
[New Thread 0x7fffe789f700 (LWP 9590)]
[New Thread 0x7fffe6dff700 (LWP 9591)]
[New Thread 0x7fffe65fe700 (LWP 9592)]
[New Thread 0x7fffdf1a3700 (LWP 9593)]
[New Thread 0x7fffd0ad0700 (LWP 9594)]
[New Thread 0x7fffc8e3a700 (LWP 9595)]
item  QDeclarativeGeoMapQuickItem(0x55555c5b5d80, parent=0x0, geometry=-17595.8,27236.7 32x32)  still referenced
item  QDeclarativeGeoMapQuickItem(0x55555c5b13c0, parent=0x0, geometry=-17602.9,27242.5 32x32)  still referenced
item  QDeclarativeGeoMapQuickItem(0x55555c5aca00, parent=0x0, geometry=-17603.3,27239.9 32x32)  still referenced
item  QDeclarativeGeoMapQuickItem(0x55555c5a8060, parent=0x0, geometry=-17602.9,27242.3 32x32)  still referenced
item  QDeclarativeGeoMapQuickItem(0x55555c5a36a0, parent=0x0, geometry=-17603.8,27239.3 32x32)  still referenced
item  QDeclarativeGeoMapQuickItem(0x55555c59ed00, parent=0x0, geometry=-17667.9,27222.3 32x32)  still referenced
item  QDeclarativeGeoMapQuickItem(0x55555c59a620, parent=0x0, geometry=-17661.3,27223.9 32x32)  still referenced
item  QDeclarativeGeoMapQuickItem(0x55555c595c80, parent=0x0, geometry=-16946.4,26722.5 32x32)  still referenced
item  QDeclarativeGeoMapQuickItem(0x55555c591310, parent=0x0, geometry=-16947.7,26723.5 32x32)  still referenced
item  QDeclarativeGeoMapQuickItem(0x55555c338980, parent=0x0, geometry=-17526.1,27234.7 32x32)  still referenced
item  QDeclarativeGeoMapQuickItem(0x55555c333fe0, parent=0x0, geometry=-16917.7,26716.4 32x32)  still referenced
item  QDeclarativeGeoMapQuickItem(0x55555c32f900, parent=0x0, geometry=-17008.5,26774.4 32x32)  still referenced
item  QDeclarativeGeoMapQuickItem(0x55555c32af40, parent=0x0, geometry=-17029.7,26777.1 32x32)  still referenced
item  QDeclarativeGeoMapQuickItem(0x55555c3185f0, parent=0x0, geometry=-17025.6,26781.7 32x32)  still referenced
item  QDeclarativeGeoMapQuickItem(0x55555bfe7a50, parent=0x0, geometry=-17020.9,26785 32x32)  still referenced
item  QDeclarativeGeoMapQuickItem(0x55555bfe30b0, parent=0x0, geometry=-17914.8,27091.1 32x32)  still referenced
[... 707 lines cut ...]
#14 0x00007ffff344811d in  () at /usr/lib/libQt5Widgets.so.5
#15 0x00007ffff344f912 in  () at /usr/lib/libQt5Widgets.so.5
#16 0x00007ffff345097c in QMenu::mouseReleaseEvent(QMouseEvent*) () at /usr/lib/libQt5Widgets.so.5
#17 0x00007ffff3313658 in QWidget::event(QEvent*) () at /usr/lib/libQt5Widgets.so.5
#18 0x00007ffff345308c in QMenu::event(QEvent*) () at /usr/lib/libQt5Widgets.so.5
#19 0x00007ffff32d3e14 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/libQt5Widgets.so.5
#20 0x00007ffff32db92a in QApplication::notify(QObject*, QEvent*) () at /usr/lib/libQt5Widgets.so.5
#21 0x00007ffff1c2e119 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/libQt5Core.so.5
#22 0x00007ffff32dac11 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) () at /usr/lib/libQt5Widgets.so.5
#23 0x00007ffff332ed57 in  () at /usr/lib/libQt5Widgets.so.5
#24 0x00007ffff33315d5 in  () at /usr/lib/libQt5Widgets.so.5
#25 0x00007ffff32d3e14 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/libQt5Widgets.so.5
#26 0x00007ffff32db6e1 in QApplication::notify(QObject*, QEvent*) () at /usr/lib/libQt5Widgets.so.5
#27 0x00007ffff1c2e119 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/libQt5Core.so.5
#28 0x00007ffff287ad04 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () at /usr/lib/libQt5Gui.so.5
#29 0x00007ffff287ccc6 in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) () at /usr/lib/libQt5Gui.so.5
#30 0x00007ffff2856a4c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/libQt5Gui.so.5
#31 0x00007fffe5a1031d in  () at /usr/lib/libQt5XcbQpa.so.5
#32 0x00007ffff1c2cdac in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/libQt5Core.so.5
#33 0x00007ffff1c350a6 in QCoreApplication::exec() () at /usr/lib/libQt5Core.so.5
#34 0x000055555569e480 in main (argc=<optimized out>, argv=<optimized out>) at /home/xxxxx/Diveapps/subsurface/subsurface-desktop-main.cpp:115
#35 0x00007ffff14ee223 in __libc_start_main () at /usr/lib/libc.so.6
#36 0x00005555556a2e0e in _start () at /home/xxxxx/Diveapps/subsurface/subsurface-desktop-main.cpp:200

### Mentions:

dirkhh, 2018-08-28:
@bstoeger has just made some changes in that code - not right where you see the crash... but it would be interesting... if you revert the last three commits (so go back to -300) and try again, does it still crash?

p-neves, 2018-08-28:
Dirk: 
I'm not sure how to revert to -300. How can I see which commits to revert? I should know more about git....

Cheers:
Pedro

p-neves, 2018-08-28:
should be this one? ec586d0e0a19cb7af62506fc2a3cb2f53198bcb8

dirkhh, 2018-08-28:
Sorry. Should have just given you the SHA: 80fe8fb331edd494ba53efb0879eb9d53031216e

dirkhh, 2018-08-28:
Thanks for testing that.
Sadly I can't reproduce either of the two crashes, even with your log file, which will make it a lot harder to fix them. But I'll see what I can do just playing with the code.

Do you have experience with `valgrind`? Often that makes it easier to spot memory access errors that end up leading to crashes later in odd locations...

p-neves, 2018-08-28:
On 8/28/18 8:27 PM, Dirk Hohndel wrote:
>
> Thanks for testing that.
> Sadly I can't reproduce either of the two crashes, even with your log 
> file, which will make it a lot harder to fix them. But I'll see what I 
> can do just playing with the code.
>
Damn...

> Do you have experience with |valgrind|? Often that makes it easier to 
> spot memory access errors that end up leading to crashes later in odd 
> locations...
>
No. But I can try.

Any pointers?

Pedro

p-neves, 2018-08-28:
Dirk:

I did valgrind ./subsurface

And this is what I got:

==27902== Memcheck, a memory error detector
==27902== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==27902== Using Valgrind-3.14.0.GIT and LibVEX; rerun with -h for 
copyright info
==27902== Command: ./subsurface
==27902==

valgrind: m_debuginfo/debuginfo.c:453 (discard_or_archive_DebugInfo): 
Assertion 'is_DebugInfo_active(di)' failed.

host stacktrace:
==27902==    at 0x58048064: show_sched_status_wrk (m_libcassert.c:369)
==27902==    by 0x58048177: report_and_quit (m_libcassert.c:440)
==27902==    by 0x58048309: vgPlain_assert_fail (m_libcassert.c:506)
==27902==    by 0x5807AD1C: discard_or_archive_DebugInfo (debuginfo.c:453)
==27902==    by 0x5807B766: discard_or_archive_marked_DebugInfos 
(debuginfo.c:598)
==27902==    by 0x5807B766: discard_DebugInfos_which_overlap_with 
(debuginfo.c:624)
==27902==    by 0x5807B766: di_notify_ACHIEVE_ACCEPT_STATE (debuginfo.c:930)
==27902==    by 0x5807B766: vgPlain_di_notify_mmap (debuginfo.c:1290)
==27902==    by 0x580AD10B: vgModuleLocal_generic_PRE_sys_mmap 
(syswrap-generic.c:2402)
==27902==    by 0x580B8BAF: vgSysWrap_amd64_linux_sys_mmap_before 
(syswrap-amd64-linux.c:404)
==27902==    by 0x580A8821: vgPlain_client_syscall (syswrap-main.c:1863)
==27902==    by 0x580A4D5A: handle_syscall (scheduler.c:1176)
==27902==    by 0x580A6616: vgPlain_scheduler (scheduler.c:1498)
==27902==    by 0x580F52E0: thread_wrapper (syswrap-linux.c:103)
==27902==    by 0x580F52E0: run_a_thread_NORETURN (syswrap-linux.c:156)

sched status:
   running_tid=1

Thread 1: status = VgTs_Runnable syscall 9 (lwpid 27902)
==27902==    at 0x401C277: mmap (in /usr/lib/ld-2.28.so)
==27902==    by 0x4007668: _dl_map_object_from_fd (in /usr/lib/ld-2.28.so)
==27902==    by 0x4009D6B: _dl_map_object (in /usr/lib/ld-2.28.so)
==27902==    by 0x401423E: dl_open_worker (in /usr/lib/ld-2.28.so)
==27902==    by 0xB2BBE76: _dl_catch_exception (in /usr/lib/libc-2.28.so)
==27902==    by 0x4013DFE: _dl_open (in /usr/lib/ld-2.28.so)
==27902==    by 0xB8C5159: ??? (in /usr/lib/libdl-2.28.so)
==27902==    by 0xB2BBE76: _dl_catch_exception (in /usr/lib/libc-2.28.so)
==27902==    by 0xB2BBF12: _dl_catch_error (in /usr/lib/libc-2.28.so)
==27902==    by 0xB8C58BE: ??? (in /usr/lib/libdl-2.28.so)
==27902==    by 0xB8C51F9: dlopen (in /usr/lib/libdl-2.28.so)
==27902==    by 0xABC1A4D: ??? (in /usr/lib/libQt5Core.so.5.11.1)
==27902==    by 0xABBB3C5: ??? (in /usr/lib/libQt5Core.so.5.11.1)
==27902==    by 0xABBB990: ??? (in /usr/lib/libQt5Core.so.5.11.1)
==27902==    by 0xABB1860: QFactoryLoader::instance(int) const (in 
/usr/lib/libQt5Core.so.5.11.1)
==27902==    by 0xA574DD6: 
QNetworkConfigurationManagerPrivate::updateConfigurations() (in 
/usr/lib/libQt5Network.so.5.11.1)
==27902==    by 0xA576C00: 
QNetworkConfigurationManagerPrivate::initialize() (in 
/usr/lib/libQt5Network.so.5.11.1)
==27902==    by 0xA56EC9E: qNetworkConfigurationManagerPrivate() (in 
/usr/lib/libQt5Network.so.5.11.1)
==27902==    by 0xA56ED38: 
QNetworkConfigurationManager::QNetworkConfigurationManager(QObject*) (in 
/usr/lib/libQt5Network.so.5.11.1)
==27902==    by 0xA4F9019: 
QNetworkAccessManager::QNetworkAccessManager(QObject*) (in 
/usr/lib/libQt5Network.so.5.11.1)
==27902==    by 0x3FDFA1: ImageDownloader::ImageDownloader() 
(imagedownloader.cpp:28)
==27902==    by 0x25615F: __static_initialization_and_destruction_0 
(imagedownloader.cpp:22)
==27902==    by 0x25615F: _GLOBAL__sub_I_imagedownloader.cpp 
(imagedownloader.cpp:516)
==27902==    by 0x4FD2BC: __libc_csu_init (in 
/home/paneves/Diveapps/subsurface/build/subsurface)
==27902==    by 0xB1A81AD: (below main) (in /usr/lib/libc-2.28.so)
client stack range: [0x1FFEFFE000 0x1FFF000FFF] client SP: 0x1FFEFFED18
valgrind stack range: [0x10031BF000 0x10032BEFFF] top usage: 9904 of 1048576

Thread 2: status = VgTs_Init (lwpid 0)
==27902==    at 0xB27FA35: clone (in /usr/lib/libc-2.28.so)
==27902==    by 0x5AF899F: ??? (in /usr/lib/libpthread-2.28.so)
==27902==    by 0x15B726FF: ???
client stack range: [0x15373000 0x15B71FFF] client SP: 0x15B71E70
valgrind stack range: [0x1007F4F000 0x100804EFFF] top usage: 312 of 1048576

Thread 3: status = VgTs_Init (lwpid 0)
==27902==    at 0xB27FA35: clone (in /usr/lib/libc-2.28.so)
==27902==    by 0x5AF899F: ??? (in /usr/lib/libpthread-2.28.so)
==27902==    by 0x166006FF: ???
client stack range: [0x15E01000 0x165FFFFF] client SP: 0x165FFE70
valgrind stack range: [0x1008053000 0x1008152FFF] top usage: 312 of 1048576

Is this ok?

Pedro

bstoeger, 2018-08-28:
If the logs don't contain sensitive data, I could try to have a look at them tomorrow.
Perhaps it is the combination of the two logs that causes the crash?

dirkhh, 2018-08-28:
That seems to be a weird assertion in the early startup of Qt. Not helpful in finding this bug, I think.
You are doing this on current ArchLinux, correct? Let me try and see if I can at least reproduce the Valgrind issue and figure out how to work around them...

bstoeger, 2018-08-28:
> After reverting to 80fe8fb it still crashes with my logbook

But the dump should look very different, as this code didn't exist before.

We seem to get a broken trip from the `DiveTripModel`...

dirkhh, 2018-08-28:
Arg. ArchLinux is on a very recent Valgrind, and that seems to have a bug that triggers the above error, see https://bugs.kde.org/show_bug.cgi?id=393146

p-neves, 2018-08-29:
I've fiddling around with my log and saw that:

1) Each time I try to remove all dives from a trip, Subsurface segfaults;

2) However, I can remove all dives except one on each trip without segfaults

3) Deleting dives takes a few seconds (inside or outside trips)

If, on a trip with only one dive, I remove that dive from the trip, I get a segfault with the following bt:

(gdb) bt
#0  0x00005555557f23b5 in MultiFilterSortModel::filterAcceptsRow(int, QModelIndex const&) const (this=<optimized out>, source_row=<optimized out>, source_parent=...) at /home/xxxxx/Diveapps/subsurface/qt-models/filtermodels.cpp:430
#1  0x00007ffff1c0809b in  () at /usr/lib/libQt5Core.so.5
#2  0x00007ffff1c0a4ea in  () at /usr/lib/libQt5Core.so.5
#3  0x00007ffff1c0a8c1 in  () at /usr/lib/libQt5Core.so.5
#4  0x00007ffff1c0fc62 in QSortFilterProxyModel::setSourceModel(QAbstractItemModel*) () at /usr/lib/libQt5Core.so.5
#5  0x000055555576781e in DiveListView::reload(DiveTripModel::Layout, bool) (this=0x5555565bacd0, layout=DiveTripModel::TREE, forceSort=<optimized out>) at /home/xxxxxx/Diveapps/subsurface/desktop-widgets/divelistview.cpp:430
#6  0x00005555556b256b in MainWindow::recreateDiveList() (this=<optimized out>) at /home/xxxxxx/Diveapps/subsurface/desktop-widgets/mainwindow.cpp:487
#7  0x00005555556b96a8 in MainWindow::refreshDisplay(bool) (this=0x555555e39fe0, doRecreateDiveList=<optimized out>) at /home/xxxxxx/Diveapps/subsurface/desktop-widgets/mainwindow.cpp:474
#8  0x0000555555728554 in UndoRemoveDivesFromTrip::redo() (this=0x7fffd4008ba0) at /home/xxxxx/Diveapps/subsurface/desktop-widgets/undocommands.cpp:171
#9  0x00007ffff3649174 in QUndoStack::push(QUndoCommand*) () at /usr/lib/libQt5Widgets.so.5
#10 0x0000555555768031 in DiveListView::removeFromTrip() (this=0x5555565bacd0) at /home/xxxxxx/Diveapps/subsurface/desktop-widgets/divelistview.cpp:691
#11 0x00007ffff1c57f5c in QMetaObject::activate(QObject*, int, int, void**) () at /usr/lib/libQt5Core.so.5
#12 0x00007ffff32cd613 in QAction::triggered(bool) () at /usr/lib/libQt5Widgets.so.5
#13 0x00007ffff32cfd0a in QAction::activate(QAction::ActionEvent) () at /usr/lib/libQt5Widgets.so.5
#14 0x00007ffff344811d in  () at /usr/lib/libQt5Widgets.so.5
#15 0x00007ffff344f912 in  () at /usr/lib/libQt5Widgets.so.5
#16 0x00007ffff345097c in QMenu::mouseReleaseEvent(QMouseEvent*) () at /usr/lib/libQt5Widgets.so.5
#17 0x00007ffff3313658 in QWidget::event(QEvent*) () at /usr/lib/libQt5Widgets.so.5
#18 0x00007ffff345308c in QMenu::event(QEvent*) () at /usr/lib/libQt5Widgets.so.5
#19 0x00007ffff32d3e14 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/libQt5Widgets.so.5
#20 0x00007ffff32db92a in QApplication::notify(QObject*, QEvent*) () at /usr/lib/libQt5Widgets.so.5
#21 0x00007ffff1c2e119 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/libQt5Core.so.5
#22 0x00007ffff32dac11 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) () at /usr/lib/libQt5Widgets.so.5
#23 0x00007ffff332ed57 in  () at /usr/lib/libQt5Widgets.so.5
#24 0x00007ffff33315d5 in  () at /usr/lib/libQt5Widgets.so.5
#25 0x00007ffff32d3e14 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/libQt5Widgets.so.5
#26 0x00007ffff32db6e1 in QApplication::notify(QObject*, QEvent*) () at /usr/lib/libQt5Widgets.so.5
#27 0x00007ffff1c2e119 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/libQt5Core.so.5
#28 0x00007ffff287ad04 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () at /usr/lib/libQt5Gui.so.5
#29 0x00007ffff287ccc6 in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) () at /usr/lib/libQt5Gui.so.5
#30 0x00007ffff2856a4c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/libQt5Gui.so.5
#31 0x00007fffe5a1031d in  () at /usr/lib/libQt5XcbQpa.so.5
#32 0x00007ffff1c2cdac in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/libQt5Core.so.5
#33 0x00007ffff344d496 in QMenu::exec(QPoint const&, QAction*) () at /usr/lib/libQt5Widgets.so.5
#34 0x000055555576bd19 in DiveListView::contextMenuEvent(QContextMenuEvent*) (this=0x5555565bacd0, event=0x7fffffffd640) at /usr/include/qt/QtGui/qevent.h:523
#35 0x00007ffff3313658 in QWidget::event(QEvent*) () at /usr/lib/libQt5Widgets.so.5
#36 0x00007ffff33b9d9f in QFrame::event(QEvent*) () at /usr/lib/libQt5Widgets.so.5
#37 0x00007ffff3534204 in QAbstractItemView::viewportEvent(QEvent*) () at /usr/lib/libQt5Widgets.so.5
#38 0x00007ffff359e0bc in QTreeView::viewportEvent(QEvent*) () at /usr/lib/libQt5Widgets.so.5
#39 0x00007ffff1c2de2b in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) () at /usr/lib/libQt5Core.so.5
#40 0x00007ffff32d3e04 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/libQt5Widgets.so.5
#41 0x00007ffff32dbb90 in QApplication::notify(QObject*, QEvent*) () at /usr/lib/libQt5Widgets.so.5
#42 0x00007ffff1c2e119 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/libQt5Core.so.5
#43 0x00007ffff332f194 in  () at /usr/lib/libQt5Widgets.so.5
#44 0x00007ffff33315d5 in  () at /usr/lib/libQt5Widgets.so.5
#45 0x00007ffff32d3e14 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/libQt5Widgets.so.5
[... cut ...]
#52 0x00007ffff1c2cdac in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/libQt5Core.so.5
#53 0x00007ffff1c350a6 in QCoreApplication::exec() () at /usr/lib/libQt5Core.so.5
#54 0x000055555569e470 in main (argc=<optimized out>, argv=<optimized out>) at /home/xxxxxx/Diveapps/subsurface/subsurface-desktop-main.cpp:115
#55 0x00007ffff14ee223 in __libc_start_main () at /usr/lib/libc.so.6
#56 0x00005555556a2dfe in _start () at /home/xxxxxx/Diveapps/subsurface/subsurface-desktop-main.cpp:200

Does this help?

bstoeger, 2018-08-29:
Unfortunately that seems like a different problem. :( Anyway, posted a PR.

p-neves, 2018-08-29:
I'm trying to narrow it down by removing trips manually....

dirkhh, 2018-08-31:
same comment as on #1608 - I just cannot recreate this :-(
I assume you are on "latest" Arch? How much memory does you system have?

bstoeger, 2018-08-31:
Could you perhaps send me a minimal dive-log that does this?
Does it also happen if no other dive-log was opened before?

janmulder, 2018-08-31:
Attached  2 dive (and 1 trip) excerpt from my logbook shows the problem for me (.txt added for Github).

[test.ssrf.txt](https://github.com/Subsurface-divelog/subsurface/files/2339390/test.ssrf.txt)

And it reproduces also when opening this minimal logbook from the start of the application.

bstoeger, 2018-08-31:
Works for me. :(

janmulder, 2018-08-31:
What Qt version are you building against? Mine is built with Qt Version 5.11.1, runtime from Qt Version 5.11.1

bstoeger, 2018-08-31:
```
$ ./subsurface -v
Subsurface v4.8.1-312-g59a68fe9b5bd,
built with libdivecomputer v0.7.0-devel-Subsurface-NG (8f4945dc1e83c53ed9d2cdbaaa16e7b117df1f32)
built with Qt Version 5.9.5, runtime from Qt Version 5.9.5
built with libgit2 0.26.0
"validateGL(): created OpenGLContext."
"validateGL(): obtained QOpenGLFunctions."
"validateGL(): detected OpenGL version 3.0."
Plugins Directory:  QDir( "/home/bs/src/subsurface/build" , nameFilters = { "*" },  QDir::SortFlags( Name | IgnoreCase ) , QDir::Filters( Dirs|Files|Drives|AllEntries ) )
qrc:/qml/MapWidget.qml:3:1: module "QtLocation" is not installed
QUrl("qrc:/qml/MapWidget.qml") failed to load with status: QQuickWidget::Status(Error)
loading dive data from ("/home/bs/src/subsurface/build/test.ssrf")
Set the current dive site: 1746586480
Set the current dive site: 1746586480
```

```
$ g++ -v                                                                                                                                               
Using built-in specs.                                                                                                                                                                  
COLLECT_GCC=g++                                                                                                                                                                        
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/7/lto-wrapper                                                                                                                        
OFFLOAD_TARGET_NAMES=nvptx-none                                                                                                                                                        
OFFLOAD_TARGET_DEFAULT=1                                                                                                                                                               
Target: x86_64-linux-gnu                                                                                                                                                               
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 7.3.0-16ubuntu3' --with-bugurl=file:///usr/share/doc/gcc-7/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --with-as=/usr/bin/x86_64-linux-gnu-as --with-ld=/usr/bin/x86_64-linux-gnu-ld --program-suffix=-7 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-libmpx --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu                                                                                                                                                                          
Thread model: posix
gcc version 7.3.0 (Ubuntu 7.3.0-16ubuntu3)
```

bstoeger, 2018-08-31:
To those who can reproduce this, could you please apply this patch and post the output:
```
diff --git a/core/divelist.c b/core/divelist.c
index f643d14f6..5e99ea125 100644
--- a/core/divelist.c
+++ b/core/divelist.c
@@ -747,6 +747,7 @@ static void delete_trip(dive_trip_t *trip)
 	dive_trip_t **p, *tmp;
 
 	assert(!trip->dives);
+	printf(" delete trip %p\n", trip);
 
 	/* Remove the trip from the list of trips */
 	p = &dive_trip_list;
@@ -822,6 +823,7 @@ void remove_dive_from_trip(struct dive *dive, short was_autogen)
 	if (!trip)
 		return;
 
+	printf(" remove dive %p from trip %p\n", dive, trip);
 	/* Remove the dive from the trip's list of dives */
 	next = dive->next;
 	pprev = dive->pprev;
@@ -843,6 +845,7 @@ void remove_dive_from_trip(struct dive *dive, short was_autogen)
 
 void add_dive_to_trip(struct dive *dive, dive_trip_t *trip)
 {
+	printf(" add dive %p to trip %p\n", dive, trip);
 	if (dive->divetrip == trip)
 		return;
 	remove_dive_from_trip(dive, false);
diff --git a/qt-models/filtermodels.cpp b/qt-models/filtermodels.cpp
index beeb7501b..f844aaa89 100644
--- a/qt-models/filtermodels.cpp
+++ b/qt-models/filtermodels.cpp
@@ -420,6 +420,7 @@ bool MultiFilterSortModel::filterAcceptsRow(int source_row, const QModelIndex &s
 
 	// Since this is not a dive, it must be a trip
 	QVariant tripVariant = sourceModel()->data(index0, DiveTripModel::TRIP_ROLE);
+	printf("tripVariant.isValid(): %d value: %p\n", tripVariant.isValid(), tripVariant.value<void *>());
 	dive_trip *trip = (dive_trip *)tripVariant.value<void *>();
 
 	if (!trip)
```
Thanks!

dirkhh, 2018-08-31:
I'm on my phone, won't get a chance to test this for a few more hours, but in my testing `.value()' returned an invalid pointer for invalid QVariant

janmulder, 2018-08-31:
@bstoeger: output of the debug print statements are:

```
add dive 0x5555567ac070 to trip 0x5555567ac000
tripVariant.isValid(): 1 value: 0x5555567ac000
tripVariant.isValid(): 1 value: 0x5555567ac000
 remove dive 0x5555567ac070 from trip 0x5555567ac000
 delete trip 0x5555567ac000
tripVariant.isValid(): 1 value: 0x5555567ac000
```
This used my minimal 2 dive logbook, and I remove the one dive from the trip.

dirkhh, 2018-08-31:
We do this weird thing where we keep the old model around and delete it later... for me the setSourceModel() in the third row from the bottom of this code is in the stack trace of the crash...
```
void DiveListView::reload(DiveTripModel::Layout layout, bool forceSort)
{
	if (layout == DiveTripModel::CURRENT)
		layout = currentLayout;
	else
		currentLayout = layout;

	header()->setSectionsClickable(true);
	connect(header(), SIGNAL(sectionPressed(int)), this, SLOT(headerClicked(int)), Qt::UniqueConnection);

	QSortFilterProxyModel *m = qobject_cast<QSortFilterProxyModel *>(model());
	QAbstractItemModel *oldModel = m->sourceModel();
	DiveTripModel *tripModel = new DiveTripModel(this);
	tripModel->setLayout(layout);
	m->setSourceModel(tripModel);
	if (oldModel)
		delete oldModel;
```

janmulder, 2018-08-31:
With the 2 extra pint lines I get: 

```
DiveTripModel::DiveTripModel()
DiveTripModel::DiveTripModel()
DiveTripModel::DiveTripModel()
DiveTripModel::DiveTripModel()
DiveTripModel::DiveTripModel()
DiveTripModel::DiveTripModel()
DiveTripModel::DiveTripModel()
DiveTripModel::DiveTripModel()
DiveTripModel::DiveTripModel()
DiveTripModel::DiveTripModel()
DiveTripModel::DiveTripModel()
DiveTripModel::DiveTripModel()
DiveTripModel::DiveTripModel()
DiveTripModel::DiveTripModel()
DiveTripModel::DiveTripModel()
DiveTripModel::DiveTripModel()
DiveTripModel::DiveTripModel()
DiveTripModel::DiveTripModel()
DiveTripModel::DiveTripModel()
DiveTripModel::setupModelData()
 add dive 0x5555567ab880 to trip 0x5555567ab810
DiveTripModel::DiveTripModel()
DiveTripModel::setupModelData()
Add dive 0x5555567ab880 in trip 0x5555567ab810 to model
Add dive 0x5555567a2010 in trip (nil) to model
tripVariant.isValid(): 1 value: 0x5555567ab810
tripVariant.isValid(): 1 value: 0x5555567ab810
 remove dive 0x5555567ab880 from trip 0x5555567ab810
 delete trip 0x5555567ab810
DiveTripModel::DiveTripModel()
DiveTripModel::setupModelData()
Add dive 0x5555567ab880 in trip (nil) to model
Add dive 0x5555567a2010 in trip (nil) to model
tripVariant.isValid(): 1 value: 0x5555567ab810
```

janmulder, 2018-08-31:
Interestingly, when I delete the oldModel right away (before the  `new DiveTripModel(this))` I cannot reproduce the crash any more. Seems a valid fix, as I do not see any reason to keep it around, but I do not understand why it fixes the issue.

bstoeger, 2018-08-31:
> Interestingly, when I delete the oldModel right away (before the new DiveTripModel(this)) I cannot reproduce the crash any more. Seems a valid fix, as I do not see any reason to keep it around, but I do not understand why it fixes the issue.

Could it be that for some reason `setSourceModel()` does access the old model? I don't see where it would do that in the source.

Clearly, the new model seems to be correctly initialized to two dives without trip:
```
Add dive 0x5555567ab880 in trip (nil) to model
Add dive 0x5555567a2010 in trip (nil) to model
```

Another thing I had in mind is that the `TreeModel` does ominous things with the `internalPointer`. See `TreeModel::data()`. What happens if an old index comes around? It will contain a pointer to an invalid item! This code is highly dubious. Perhaps the deletion indirectly invalidates all the indexes?

Could you try one last printf thing (on top of fresh master):
```
diff --git a/core/divelist.c b/core/divelist.c
index f643d14f6..5e99ea125 100644
--- a/core/divelist.c
+++ b/core/divelist.c
@@ -747,6 +747,7 @@ static void delete_trip(dive_trip_t *trip)
 	dive_trip_t **p, *tmp;
 
 	assert(!trip->dives);
+	printf(" delete trip %p\n", trip);
 
 	/* Remove the trip from the list of trips */
 	p = &dive_trip_list;
@@ -822,6 +823,7 @@ void remove_dive_from_trip(struct dive *dive, short was_autogen)
 	if (!trip)
 		return;
 
+	printf(" remove dive %p from trip %p\n", dive, trip);
 	/* Remove the dive from the trip's list of dives */
 	next = dive->next;
 	pprev = dive->pprev;
@@ -843,6 +845,7 @@ void remove_dive_from_trip(struct dive *dive, short was_autogen)
 
 void add_dive_to_trip(struct dive *dive, dive_trip_t *trip)
 {
+	printf(" add dive %p to trip %p\n", dive, trip);
 	if (dive->divetrip == trip)
 		return;
 	remove_dive_from_trip(dive, false);
diff --git a/qt-models/divetripmodel.cpp b/qt-models/divetripmodel.cpp
index 384da6b89..4b98d174d 100644
--- a/qt-models/divetripmodel.cpp
+++ b/qt-models/divetripmodel.cpp
@@ -441,6 +441,7 @@ DiveTripModel::DiveTripModel(QObject *parent) :
 	TreeModel(parent),
 	currentLayout(TREE)
 {
+	printf("DiveTripModel::DiveTripModel()\n");
 	columns = COLUMNS;
 }
 
@@ -582,6 +583,7 @@ QVariant DiveTripModel::headerData(int section, Qt::Orientation orientation, int
 
 void DiveTripModel::setupModelData()
 {
+	printf("DiveTripModel::setupModelData(): %p\n", this);
 	int i = dive_table.nr;
 
 	beginResetModel();
@@ -593,6 +595,7 @@ void DiveTripModel::setupModelData()
 		struct dive *dive = get_dive(i);
 		update_cylinder_related_info(dive);
 		dive_trip_t *trip = dive->divetrip;
+		printf("Add dive %p in trip %p to model\n", dive, trip);
 
 		DiveItem *diveItem = new DiveItem();
 		diveItem->diveId = dive->id;
diff --git a/qt-models/filtermodels.cpp b/qt-models/filtermodels.cpp
index beeb7501b..fc01cea0c 100644
--- a/qt-models/filtermodels.cpp
+++ b/qt-models/filtermodels.cpp
@@ -410,6 +410,7 @@ bool MultiFilterSortModel::showDive(const struct dive *d) const
 
 bool MultiFilterSortModel::filterAcceptsRow(int source_row, const QModelIndex &source_parent) const
 {
+	printf("MultiFilterSortModel::filterAcceptsRow() sourceModel(): %p\n", sourceModel());
 	QModelIndex index0 = sourceModel()->index(source_row, 0, source_parent);
 	QVariant diveVariant = sourceModel()->data(index0, DiveTripModel::DIVE_ROLE);
 	struct dive *d = (struct dive *)diveVariant.value<void *>();
@@ -420,6 +421,7 @@ bool MultiFilterSortModel::filterAcceptsRow(int source_row, const QModelIndex &s
 
 	// Since this is not a dive, it must be a trip
 	QVariant tripVariant = sourceModel()->data(index0, DiveTripModel::TRIP_ROLE);
+	printf("tripVariant.isValid(): %d value: %p\n", tripVariant.isValid(), tripVariant.value<void *>());
 	dive_trip *trip = (dive_trip *)tripVariant.value<void *>();
 
 	if (!trip)
diff --git a/qt-models/treemodel.cpp b/qt-models/treemodel.cpp
index 5ddd2d3a2..aae90cf0a 100644
--- a/qt-models/treemodel.cpp
+++ b/qt-models/treemodel.cpp
@@ -46,6 +46,7 @@ QVariant TreeModel::data(const QModelIndex &index, int role) const
 		return QVariant();
 
 	TreeItem *item = static_cast<TreeItem *>(index.internalPointer());
+	printf("\tTreeModel::data(): %p index: %p row: %d item: %p\n", this, &index, index.row(), item);
 	QVariant val = item->data(index.column(), role);
 
 	if (role == Qt::FontRole && !val.isValid())
```

This will produce tons of output, but you probably can reduce it by running it through `uniq`.

janmulder, 2018-08-31:
Output upto the crash, reduced by `uniq`

```
DiveTripModel::DiveTripModel()
DiveTripModel::setupModelData(): 0x5555566de350
 add dive 0x5555567abd40 to trip 0x5555567abcd0
DiveTripModel::DiveTripModel()
DiveTripModel::setupModelData(): 0x555555dd90a0
Add dive 0x5555567abd40 in trip 0x5555567abcd0 to model
Add dive 0x5555567a24d0 in trip (nil) to model
MultiFilterSortModel::filterAcceptsRow() sourceModel(): 0x555555dd90a0
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffd730 row: 0 item: 0x55555679fc90
tripVariant.isValid(): 1 value: 0x5555567abcd0
MultiFilterSortModel::filterAcceptsRow() sourceModel(): 0x555555dd90a0
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffd730 row: 1 item: 0x555556799c50
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffd4b0 row: 0 item: 0x55555679fc90
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffd490 row: 1 item: 0x555556799c50
MultiFilterSortModel::filterAcceptsRow() sourceModel(): 0x555555dd90a0
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffd790 row: 0 item: 0x5555568213f0
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffd560 row: 0 item: 0x5555568213f0
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffd220 row: 0 item: 0x5555568213f0
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffcda0 row: 1 item: 0x555556799c50
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffccc0 row: 1 item: 0x555556799c50
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffced0 row: 1 item: 0x555556799c50
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffcda0 row: 1 item: 0x555556799c50
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffcda0 row: 0 item: 0x5555568213f0
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffccc0 row: 0 item: 0x5555568213f0
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffced0 row: 0 item: 0x5555568213f0
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffcda0 row: 0 item: 0x5555568213f0
MultiFilterSortModel::filterAcceptsRow() sourceModel(): 0x555555dd90a0
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffd930 row: 0 item: 0x55555679fc90
tripVariant.isValid(): 1 value: 0x5555567abcd0
MultiFilterSortModel::filterAcceptsRow() sourceModel(): 0x555555dd90a0
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffd930 row: 1 item: 0x555556799c50
MultiFilterSortModel::filterAcceptsRow() sourceModel(): 0x555555dd90a0
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffd890 row: 0 item: 0x5555568213f0
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffda20 row: 0 item: 0x5555568213f0
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffd7b0 row: 0 item: 0x55555679fc90
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffd640 row: 0 item: 0x5555568213f0
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffd7b0 row: 1 item: 0x555556799c50
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffd870 row: 0 item: 0x55555679fc90
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffd700 row: 0 item: 0x5555568213f0
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffd870 row: 1 item: 0x555556799c50
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffd6a0 row: 0 item: 0x5555568213f0
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffad30 row: 0 item: 0x55555679fc90
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffae60 row: 0 item: 0x55555679fc90
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffa760 row: 0 item: 0x55555679fc90
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffa890 row: 0 item: 0x55555679fc90
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffa9c0 row: 0 item: 0x55555679fc90
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffac00 row: 0 item: 0x55555679fc90
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffac00 row: 0 item: 0x5555568213f0
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffab20 row: 0 item: 0x5555568213f0
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffad30 row: 0 item: 0x5555568213f0
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffac00 row: 0 item: 0x5555568213f0
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffac00 row: 1 item: 0x555556799c50
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffab20 row: 1 item: 0x555556799c50
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffad30 row: 1 item: 0x555556799c50
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffac00 row: 1 item: 0x555556799c50
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffae60 row: 0 item: 0x55555679fc90
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffae60 row: 0 item: 0x5555568213f0
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffad80 row: 0 item: 0x5555568213f0
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffaf90 row: 0 item: 0x5555568213f0
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffae60 row: 0 item: 0x5555568213f0
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffae60 row: 1 item: 0x555556799c50
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffad80 row: 1 item: 0x555556799c50
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffaf90 row: 1 item: 0x555556799c50
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffae60 row: 1 item: 0x555556799c50
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffc410 row: 1 item: 0x555556799c50
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffc330 row: 1 item: 0x555556799c50
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffc540 row: 1 item: 0x555556799c50
[... 17 lines cut ...]
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffce00 row: 0 item: 0x55555679fc90
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffce00 row: 1 item: 0x555556799c50
[... cut ...]
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffb690 row: 0 item: 0x5555568213f0
 remove dive 0x5555567abd40 from trip 0x5555567abcd0
 delete trip 0x5555567abcd0
DiveTripModel::DiveTripModel()
DiveTripModel::setupModelData(): 0x555556b3afa0
Add dive 0x5555567abd40 in trip (nil) to model
Add dive 0x5555567a24d0 in trip (nil) to model
MultiFilterSortModel::filterAcceptsRow() sourceModel(): 0x555555dd90a0
 TreeModel::data(): 0x555555dd90a0 index: 0x7fffffffbca0 row: 0 item: 0x55555679fc90
tripVariant.isValid(): 1 value: 0x5555567abcd0
```

dirkhh, 2018-08-31:
Again on my phone.
Deleting the old model wasn't enough to fix the crash in my case.
This does smell like a QT bug to me...

bstoeger, 2018-08-31:
> This does smell like a QT bug to me...

I wouldn't be so fast. The whole `DiveTripModel` handling is a bit ominous. Matter of fact, the C-core and `DiveTripModel` go out of sync and I'm not sure that's a valid situation.

@janmulder: If you got time on your hands, could you perhaps try #1528? But no hurry.

bstoeger, 2018-08-31:
As a first step to clean this mess I propose #1617. Not sure if it fixes the bug, but at least we shouldn't have an old model flying around! Hopefully that clears things up.

janmulder, 2018-09-01:
@bstoeger: happy to try #1528, after merging of #1617 and rebasing it on the current master.

thiagomacieira, 2018-09-01:
`setSourceModel` does access the previous model if it is not null. All the implementations and the base implementation disconnect some signals from the old model before setting up the new one.

bstoeger, 2018-09-01:
Sure, but that's not what I meant.

What we are observing is that it accesses the *data* of the old model. I.e. the filters are re-evaluated for the old model, which in our case means accessing data that don't exist anymore. Unfortunately this doesn't seem to happen with my version of Qt (5.9), and therefore I don't know what the call-chain is.

Certainly, the problem in our case is that we have some fundamental layer-violations going on (UI directly accessing core [core and model are now out-of-sync] and only then resetting the model [core and model back in sync]). Still, I'm not sure if the behavior of Qt here is to be expected..? Or maybe I'm misinterpreting things?

janmulder, 2018-09-04:
@p-neves: can you still reproduce this in the current master?

p-neves, 2018-09-07:
On 9/4/18 2:05 PM, Jan Mulder wrote:
>
> @p-neves <https://github.com/p-neves>: can you still reproduce this in 
> the current master?
>

Sorry for the delay.

It seems the problem is gone in the current master, as I cannot 
reproduce it.

Cheers:

Pedro

janmulder, 2018-09-07:
Thanks for the update Pedro. Closing this.

closed 2018-09-07 by janmulder