subsurface/subsurface #1608: Subsurface segfaults when opening preferences (Related to  #1401?)

p-neves, 2018-08-28:
### Describe the issue:
Subsurface segfaults when opening preferences

- [ X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When opening program preferences, Subsurface segfaults

### Operating system:

Arch Linux

### Subsurface version:

4.8.1

Yes. With Subsurface-4.7.8-x86_64.AppImage

Self compiled from master

4.8.1-303-g66aeaddd0f9c

### Steps to reproduce:

1) Start Subsurface
2) Select File/Preferences

### Current behavior:

Subsurface segfaults

### Expected behavior:

Open application preferences

### Additional information:

It is probably related to my logbook (it's getting huge!). Please see issue #1401.

Backtrace follows:

gdb bt:

#0  0x00007ffff1629667 in __strlen_avx2 () at /usr/lib/libc.so.6
#1  0x000055555593a457 in QString::QString(char const*) (ch=0xffffffff <error: Cannot access memory at address 0xffffffff>, this=0x7fffffffcdc0) at /usr/include/qt/QtCore/qstring.h:693
#2  0x000055555593a457 in PreferencesDefaults::refreshSettings() (this=0x5555565af1d0) at /home/xxx/Diveapps/subsurface/desktop-widgets/preferences/preferences_defaults.cpp:72
#3  0x000055555592f0fd in PreferencesDialog::refreshPages() (this=0x55555622aba0) at /home/xxxxx/Diveapps/subsurface/desktop-widgets/preferences/preferencesdialog.cpp:111
#4  0x00005555556b0661 in MainWindow::on_actionPreferences_triggered() (this=<optimized out>) at /home/paneves/Diveapps/subsurface/desktop-widgets/mainwindow.cpp:814
#5  0x000055555575782a in MainWindow::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) (_o=0x555555e245e0, _c=QMetaObject::InvokeMetaMethod, _id=12, _a=0x7fffffffd030)
    at /home/paneves/Diveapps/subsurface/build/desktop-widgets/subsurface_interface_autogen/EWIEGA46WW/moc_mainwindow.cpp:356
#6  0x000055555575c3c3 in MainWindow::qt_metacall(QMetaObject::Call, int, void**) (this=0x555555e245e0, _c=QMetaObject::InvokeMetaMethod, _id=12, _a=0x7fffffffd030)
    at /home/paneves/Diveapps/subsurface/build/desktop-widgets/subsurface_interface_autogen/EWIEGA46WW/moc_mainwindow.cpp:463
#7  0x00007ffff1c5802d in QMetaObject::activate(QObject*, int, int, void**) () at /usr/lib/libQt5Core.so.5
#8  0x00007ffff32cd613 in QAction::triggered(bool) () at /usr/lib/libQt5Widgets.so.5
#9  0x00007ffff32cfd0a in QAction::activate(QAction::ActionEvent) () at /usr/lib/libQt5Widgets.so.5
#10 0x00007ffff344811d in  () at /usr/lib/libQt5Widgets.so.5
#11 0x00007ffff344f912 in  () at /usr/lib/libQt5Widgets.so.5
#12 0x00007ffff345097c in QMenu::mouseReleaseEvent(QMouseEvent*) () at /usr/lib/libQt5Widgets.so.5
#13 0x00007ffff3313658 in QWidget::event(QEvent*) () at /usr/lib/libQt5Widgets.so.5
#14 0x00007ffff345308c in QMenu::event(QEvent*) () at /usr/lib/libQt5Widgets.so.5
#15 0x00007ffff32d3e14 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/libQt5Widgets.so.5
#16 0x00007ffff32db92a in QApplication::notify(QObject*, QEvent*) () at /usr/lib/libQt5Widgets.so.5
#17 0x00007ffff1c2e119 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/libQt5Core.so.5
#18 0x00007ffff32dac11 in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool) () at /usr/lib/libQt5Widgets.so.5
#19 0x00007ffff332ed57 in  () at /usr/lib/libQt5Widgets.so.5
#20 0x00007ffff33315d5 in  () at /usr/lib/libQt5Widgets.so.5
#21 0x00007ffff32d3e14 in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /usr/lib/libQt5Widgets.so.5
#22 0x00007ffff32db6e1 in QApplication::notify(QObject*, QEvent*) () at /usr/lib/libQt5Widgets.so.5
#23 0x00007ffff1c2e119 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /usr/lib/libQt5Core.so.5
#24 0x00007ffff287ad04 in QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) () at /usr/lib/libQt5Gui.so.5
#25 0x00007ffff287ccc6 in QGuiApplicationPrivate::processWindowSystemEvent(QWindowSystemInterfacePrivate::WindowSystemEvent*) () at /usr/lib/libQt5Gui.so.5
#26 0x00007ffff2856a4c in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/libQt5Gui.so.5
#27 0x00007fffe5a1031d in  () at /usr/lib/libQt5XcbQpa.so.5
#28 0x00007ffff1c2cdac in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /usr/lib/libQt5Core.so.5
#29 0x00007ffff1c350a6 in QCoreApplication::exec() () at /usr/lib/libQt5Core.so.5
#30 0x000055555569e480 in main (argc=<optimized out>, argv=<optimized out>) at /home/xxxxx/Diveapps/subsurface/subsurface-desktop-main.cpp:115
#31 0x00007ffff14ee223 in __libc_start_main () at /usr/lib/libc.so.6
#32 0x00005555556a2e0e in _start () at /home/xxxxx/Diveapps/subsurface/subsurface-desktop-main.cpp:200

### Mentions:

dirkhh, 2018-08-28:
I don't think this is related to your logbook. BTW: how big is it (i.e., how many dives)?
The stack trace indicates that Qt is trying to get a `strlen()` from an invalid pointer.
Specifically, it appears to happen here:
```
        for (int i = 0; tank_info[i].name != NULL && i < MAX_TANK_INFO; i++) {
                ui->default_cylinder->addItem(tank_info[i].name);
```
So my guess is that somehow tank_info[i].name is a bogus pointer (`0xffffffff`, based on the stack trace). That, of course, is worrying me, because that code hasn't been touched in a long time and I'm not quite sure how you'd end up with bogus data in there.

Of course I can't reproduce this here on ArchLinux, which makes it harder to track down.

To go back to your suspicion that this is driven by your dive log, if you open Subsurface with `--user=test` (which means you don't open your dive log and start with default settings), does it crash as well?

p-neves, 2018-08-28:
Hi Dirk:
My logbook has around 2000 dives but it's nearly 51 Mb! (I have lots of dives with data from more than one device...).

When starting subsurface with --user=test it does not crash. 

Cheers:

Pedro

dirkhh, 2018-08-28:
Would you be willing to share it with me offline (compress it, send by email)?

p-neves, 2018-08-28:
Of course... I've just sent it...

dirkhh, 2018-08-31:
I cannot reproduce this with your data file, nor with the much bigger anonymized dive data from here https://github.com/Subsurface-divelog/large-anonymous-sample-data 
Not on ArchLinux with 8G, not on a Mac.
It really does feel like some sort of memory corruption, but I'm not sure how to debug it if I can't reproduce it. :-(

dirkhh, 2018-09-05:
@p-neves any update on this? Is this fixed now?

bstoeger, 2018-09-05:
> @p-neves any update on this? Is this fixed now?

I think this is not identical to #1607. The latter could be explained by the FilterModel-oddity. I don't see how that could also cause the memory-corruption(?) of this one.

p-neves, 2018-09-11:
Sorry for the delay. It still segfaults with my logbook with the current master

exxus, 2018-10-19:
I can reproduce the issue with the large anonymous-sample-data

on archlinux with current master cc225a44f2342d343e97a58b38b2f510ba1d173c
on ryzen 1700x

```
./subsurface --user=test -v --import ../../../large-anonymous-sample-data/large-anon.ssrf 
Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
Subsurface v4.8.3-213-gcc225a44f234,
built with libdivecomputer v0.7.0-devel-Subsurface-NG (e61b7c64f998efe00c2f10c7d84fa1618ded598c)
built with Qt Version 5.11.2, runtime from Qt Version 5.11.2
built with libgit2 0.27.5
"validateGL(): created OpenGLContext."
"validateGL(): obtained QOpenGLFunctions."
"validateGL(): detected OpenGL version 4.4."
using custom config for "Subsurface-test"
can't find Qt localization for locale "de-DE" searching in "/usr/share/qt/translations"
Plugins Directory:  QDir( "/home/exus/builds/subsurface/subsurface/build" , nameFilters = { "*" },  QDir::SortFlags( Name | IgnoreCase ) , QDir::Filters( Dirs|Files|Drives|AllEntries ) )
loading dive data from ("/home/exus/.subsurface/exus.xml")
Set the current dive site: 3729214962
Set the current dive site: 3729214962
importing dive data from ("../../../large-anonymous-sample-data/large-anon.ssrf")
Set the current dive site: 2774378797
Set the current dive site: 3729214962

File locations:

Unable to get local git directory
Cloud URL: No valid cloud credentials set.

Image filename table: /home/exus/.subsurface/hashes
Local picture directory: /home/exus/.subsurface/picturedata/

item  QDeclarativeGeoMapQuickItem(0x5605c6493e70, parent=0x0, geometry=11094.7,5534.84 32x32)  still referenced
item  QDeclarativeGeoMapQuickItem(0x5605c63d99c0, parent=0x0, geometry=10149,6279.51 32x32)  still referenced
.
.
.
item  QDeclarativeGeoMapQuickItem(0x5605c6fb1ce0, parent=0x0, geometry=-9273.35,19909.7 32x32)  still referenced
item  QDeclarativeGeoMapQuickItem(0x5605c736a490, parent=0x0, geometry=-8036.33,21891.4 32x32)  still referenced
[1]    23627 segmentation fault (core dumped)  ./subsurface --user=test -v --import 
./subsurface --user=test -v --import   14,08s user 3,37s system 49% cpu 35,004 total
```