subsurface/subsurface #1609: Desktop-window size too small on startup

bstoeger, 2018-08-28:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

On startup I get a very small window (with respect to the size of my desktop), which is annoying. An old instance produces a reasonably sized window, so it's not something with my settings. I have faint memories of messing with the `appW`, `appH` thing, but I can't figure out how the window-dimensions are set. I guess they are set indirectly!?
 
### Operating system:

Kubuntu 18.4

### Subsurface version:

Fails:
Subsurface v4.8.1-305-gc2d11d27aac0,
built with libdivecomputer v0.7.0-devel-Subsurface-NG (8f4945dc1e83c53ed9d2cdbaaa16e7b117df1f32)
built with Qt Version 5.9.5, runtime from Qt Version 5.9.5
built with libgit2 0.26.0
"validateGL(): created OpenGLContext."
"validateGL(): obtained QOpenGLFunctions."
"validateGL(): detected OpenGL version 3.0."

Works:
Subsurface v4.8.1-35-gb53672507dea,
built with libdivecomputer v0.7.0-devel-Subsurface-NG (02560a7e7fe82919d584d3edbf3876f90382052c)
built with Qt Version 5.9.5, runtime from Qt Version 5.9.5
built with libgit2 0.26.0
"validateGL(): created OpenGLContext."
"validateGL(): obtained QOpenGLFunctions."
"validateGL(): detected OpenGL version 3.0."

### Steps to reproduce:

Start application.

### Current behavior:

Way too small main-window.

### Expected behavior:

Appropriately sized main-window.

bstoeger, 2018-08-29:
`git bisect`ed it down to
```
commit 321a920a9873a3828a24c1b28cf8eb5fe1bff2cb
Author: jan Iversen <jani@apache.org>
Date:   Mon Aug 20 11:19:49 2018 +0200

    desktop-widgets: remove QSettings from desktop-widgets
```
Reverting this commit makes it work again. I'm a bit annoyed at myself for not noting and complaining about these move `QSettings` to preferences commits before. :(

dirkhh, 2018-08-29:
The amount of snark is entertaining, yet doesn't necessarily feel very encouraging or helpful.

dirkhh, 2018-08-29:
https://www.urbandictionary.com/define.php?term=snark
slang / contraction for a "snide remark"
your comment reads like saying "I always said this was wrong, no one listened, now see where this got us"
Oder auf Deutsch - der Kommentar klingt besserwisserisch (ha, ist das ein Wort?). Denn Du hast bekanntlich gegen diese Commits argumentiert und ich habe sie trotzdem gemerged da mir entgangen ist, dass das Laden von Disk in diesem Fall nicht automatisch getriggered wird. ("gemerged" und "getriggered" klingt auch nicht besonders Deutsch...)
Egal. Es ist 3:30am, und im Gegensatz zu Lubomir ist das bei mir eventuell nicht die beste Zeit um entspannt an Loesungen zu arbeiten...

closed 2018-08-29 by dirkhh

bstoeger, 2018-08-29:
> I always said this was wrong, no one listened, now see where this got us

If that was my intent I would have said it pretty much like that and would have posted a revert-PR.

I *do* think that these patches are a pessimization and make future work significantly harder, but this is unrelated to the bug here. And I'm still unhappy about myself for not having had a closer look before.

henrik242, 2018-08-29:
All that aside, I really hope @janiversen comes back to us.  The work he did on the settings cleanup was awesome.

janiversen, 2018-08-29:
@henrik242 thanks for your POSITIVE comment, nice to read as contrast to the current tendency.

@dirkhh I will have a look at that commit later today or tomorrow. If it is disturbing or disturbs “someone” then please revert it! and I will make a new PR. Ps I do have an idea ehat the problem might be.

janiversen, 2018-08-29:
@bstoeger, comments aside, thanks for bisecting this, and isolating my commit. I am compiling while I write this.

dirkhh, 2018-08-29:
The one commit causing @bstoeger 's reproducible issue has been reverted (that's how this issue got closed). As I said in my email to you, @janiversen , I a quick hack showed me that load_geometry() never got called - nor did the other load_xxx() functions that were created.
So while the README explains how things _should_ work, you need to update that one commit (or add a second one) and create a new PR in order to get this functionality back, once it works.

Our policy is typically to revert things right away if they demonstrably break something - while we don't have a lot of people actively using Subsurface every day based on latest master, there are enough users compiling from source that I have a strong goal of keeping master always functional.

commit 2d87a657 (2018-08-29): Revert "desktop-widgets: remove QSettings from desktop-widgets"

This reverts commit 321a920a9873a3828a24c1b28cf8eb5fe1bff2cb.

It appears that the load_xxx functions aren't called, so while the correct
values are stored to the settings, they aren't retrieved. Let's revert while
this gets fixed.

Fixes #1609