subsurface/subsurface #1620: Old warnings when compiler for osx.

janiversen, 2018-09-01:
### Describe the issue:

- [ x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When compiling for Mac, we continue getting these warnings:
## cd /Users/jani/LO/mac ##
Last login: Fri Aug 31 21:10:15 on ttys002
janMac:~ jani$ source ~/.bash_profile
janMac:~ jani$ cd develop/src
janMac:src jani$ ls
/Users/jani/develop/src/subsurface/core/color.cpp:9:13: warning: suggest braces around
      initialization of subobject [-Wmissing-braces]
        { SAC_1, { FUNGREEN1, BLACK1_LOW_TRANS } },
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~
                   {                          }
/Users/jani/develop/src/subsurface/core/color.h:12:19: note: expanded from macro 'FUNGREEN1'
#define FUNGREEN1 QColor::fromRgbF(0.0, 0.4, 0.2, 1)
                  ^
/Users/jani/develop/src/subsurface/core/color.cpp:10:13: warning: suggest braces around
      initialization of subobject [-Wmissing-braces]
        { SAC_2, { APPLE1, BLACK1_LOW_TRANS } },
                   ^~~~~~~~~~~~~~~~~~~~~~~~
                   {                       }
/Users/jani/develop/src/subsurface/core/color.cpp:11:13: warning: suggest braces around
      initialization of subobject [-Wmissing-braces]
        { SAC_3, { ATLANTIS1, BLACK1_LOW_TRANS } },
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~
                   {                          }
/Users/jani/develop/src/subsurface/core/color.h:19:19: note: expanded from macro 'ATLANTIS1'
#define ATLANTIS1 QColor::fromRgbF(0.4, 0.8, 0.2, 1)
                  ^
/Users/jani/develop/src/subsurface/core/color.cpp:12:13: warning: suggest braces around
      initialization of subobject [-Wmissing-braces]
        { SAC_4, { ATLANTIS2, BLACK1_LOW_TRANS } },
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~
                   {                          }
/Users/jani/develop/src/subsurface/core/color.h:20:19: note: expanded from macro 'ATLANTIS2'
#define ATLANTIS2 QColor::fromRgbF(0.6, 0.8, 0.2, 1)
                  ^
/Users/jani/develop/src/subsurface/core/color.cpp:13:13: warning: suggest braces around
      initialization of subobject [-Wmissing-braces]
        { SAC_5, { EARLSGREEN1, BLACK1_LOW_TRANS } },
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                   {                            }
/Users/jani/develop/src/subsurface/core/color.h:22:21: note: expanded from macro 'EARLSGREEN1'
#define EARLSGREEN1 QColor::fromRgbF(0.8, 0.8, 0.2, 1)
                    ^
/Users/jani/develop/src/subsurface/core/color.cpp:14:13: warning: suggest braces around
      initialization of subobject [-Wmissing-braces]
        { SAC_6, { HOKEYPOKEY1, BLACK1_LOW_TRANS } },
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                   {                            }
/Users/jani/develop/src/subsurface/core/color.h:31:21: note: expanded from macro 'HOKEYPOKEY1'
#define HOKEYPOKEY1 QColor::fromRgbF(0.8, 0.6, 0.2, 1)
                    ^
/Users/jani/develop/src/subsurface/core/color.cpp:15:13: warning: suggest braces around
      initialization of subobject [-Wmissing-braces]
        { SAC_7, { TUSCANY1, BLACK1_LOW_TRANS } },
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~
                   {                         }
/Users/jani/develop/src/subsurface/core/color.h:28:18: note: expanded from macro 'TUSCANY1'
#define TUSCANY1 QColor::fromRgbF(0.8, 0.4, 0.2, 1)
                 ^
/Users/jani/develop/src/subsurface/core/color.cpp:16:13: warning: suggest braces around
      initialization of subobject [-Wmissing-braces]
[... 407 lines cut ...]
/Users/jani/develop/src/subsurface/core/color.h:43:28: note: expanded from macro
      'TUNDORA1_MED_TRANS'
#define TUNDORA1_MED_TRANS QColor::fromRgbF(0.3, 0.3, 0.3, 0.5)
                           ^
57 warnings generated.

Should we not add the proposed braces ?

### Operating system:

osx newest, xcode newest.

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

no warnings.

### Additional information:

### Mentions:

dirkhh, 2018-09-01:
These are the result of the breadth of compilers that we support.
Adding the braces that clang asks for results in gcc on a couple of platforms failing to compile these files.

bstoeger, 2018-09-01:
> Feel free to try it to submit a PR and then observe the gcc output vs clang output.

I'd love to, but it seems to me that the Travis Mac build fails to compile `libdivecomputer`...?

bstoeger, 2018-09-01:
Must have been a fluke.

> [ 26%] Building CXX object core/CMakeFiles/subsurface_corelib.dir/color.cpp.o
[ 26%] Building CXX object qt-models/CMakeFiles/subsurface_models_desktop.dir/divelocationmodel.cpp.o

Looks good, no?

dirkhh, 2018-09-01:
So I think this should be fixed now. Thanks @bstoeger

closed 2018-09-01 by dirkhh