subsurface/subsurface #1623: Preferences: final sync takes forever

janmulder, 2018-09-01:
### Describe the issue:

- [X] Bug ?
- [ ] Change request
- [ ] New feature request
- [x] Discussion request

### Issue long description:
Since the new preferences code, after exit of the UI (using the menu-quit), it takes about 15 seconds before my `bash` prompt is shown. Investigation shows that a lot of time is spend in the final preferences sync (`qPref::sync();`). My Arch system does not have a working valgrind at the moment (bug in valgrind) so further investigation where those 15 seconds are spend is not easy. 

### Operating system:
OS: Arch Linux (upto date)

### Subsurface version:
What version of Subsurface are you running? master
Does the same happen on another Subsurface version?: no, seems introduces in the big preferences rewrite

### Steps to reproduce:
Start Subsurface, quit, and check the time the command prompt returns

### Current behavior:
15 seconds spend, `top` shows no relevant cpu usage, so it feels like a 15 second stall.

### Expected behavior:
Minimal delay between UI exit and process end. Writing some preferences to a simple ASCII file cannot be that hard.

@janiversen

janmulder, 2018-09-01:
Noting in the standard app logging. Nothing relevant with `dmesg` or system journal. Just commenting that final sync gives me a more or less instantaneous exit, activating the final sync, and the 15 sec delay is back.

dirkhh, 2018-09-01:
That's wild. Is anyone else seeing this? I run this on three computers and in another four VMs and haven't seen anything like this...
Which version of Qt do you build against? Anything else that's interesting or unusual about your system?

janmulder, 2018-09-01:
Agree, its very weird. Using Qt 5.11.1 (the one that comes from Arch Linux pacman). And there is noting else strange about the system.

dirkhh, 2018-09-01:
I'm running it on ArchLinux as well (but in a VM) and have not seen this problem at all. I just tried again with latest master... no problem. I hate issues like that :-(

janiversen, 2018-09-01:
I have also not seen anything nearly as bad, BUT the current implementation do have a major weakness (even though it is far better than the old code, where you could set a preference, that was never saved to disk).

Because we do not know when the app code changes structure preferences directly, I choose to save all variables when existing in qPref::sync() this is a BIG overkill, but was easy to implement :-)

I expect to have a patch in max. a day that keeps a shadow variable, and thus can see if the variable in struct preferences is changed, this will limit the actual disk writing to typically max. a few variables. Be aware this is the first step in guarding the preference variable.

When we guard the preference variables, they will only be accessible through get/set functions, and thus there are no longer a need for a qPref::sync() when the app closes. My pain in the moment is a few variables that are used in the c-code.

I hope @janmulder will test my PR when submitted.

janiversen, 2018-09-02:
When closing thee app, I see a ton of:
item  QDeclarativeGeoMapQuickItem(0x7fce968c0120, parent=0x0, geometry=-3903.47,24310.4 32x32)  still referenced

Not sure what caused this (compiled app on master from yesterday), but this could potentially be a problem ?

dirkhh, 2018-09-02:
That's been there ever since I switched to Qt 5.11 - something in the Map code is broken

janmulder, 2018-09-07:
Small update: tested this with an AppImage built on Travis today. Same delay on exit of the app. Tried a new .conf file, tried a logbook with 1 dive, tried xml and git storage: same delay. The mystery is growing :-(

janmulder, 2018-09-07:
Like I said in #1626. That worked (as in, fixed the slow shutdown of the process). But as we all know, that ended (again) in some discussion over the way to go (and I'm not going to participate in that), and that #1626 ended up closed without merge.

A more important thing is ... why am I seeing this and nobody else? That is what disturbs me way more than the actual workaround in #1626.

janmulder, 2018-09-10:
Close as fixed in commit e9e9f4c.

closed 2018-09-10 by janmulder