subsurface/subsurface #1624: opening a log file makes subsurface 4.8 use CPU100% and hang up

antiker1, 2018-09-02:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
I'm running subsurface 4.8.0-1~xenial on a Debian stable (with some older libssl as the subsurface packages do work with current Debians)...
Opening my version="3" logfile makes the CPU go to 100% and the process of opening the log never finishing. Eventually I have to kill subsurface process.

It works fine again when going back to 4.7.6-1~zesty.

### Operating system:
Debian stable
OSTC4

### Subsurface version:
4.8.0-1~xenial

Works again with 4.7.6-1~zesty.

The "official" release from ppa.launchpad.net/subsurface/subsurface/ubuntu

Any ideas on how to fix that?

dirkhh, 2018-09-02:
Most likely this is triggered by something in your log file. Would you be willing to share that with one of the developers?

antiker1, 2018-09-02:
Uhm... is there something like a logbook anonymiser script, which removes all the names/notes and that like?

mturkia, 2018-09-02:
we do not have any anonymizers. Thus the wording of sharing with one of the developers, not the public msg board. You could send it directly to me or dirk...

dirkhh, 2018-09-02:
I have a script that needs a little more work but that does an ok job with that. I'll try to get this to the point that I can post it, but not tonight...

atdotde, 2018-09-02:
Let’s make this an export option. I would suggest doing s/[a-zA-Z]/x/g on the relevant fields in the xml keeping special characters. This of course does not work well for non-Latin character languages. 

I looked briefly at the source and wondered about the best way to implement this: handing down a bool anonymize through the functions in save-xml.c or via xslt afterwards?

dirkhh, 2018-09-02:
Actually - as I thought about this some more this morning - in order to reproduce the hang we almost certainly need the unedited file. My guess is that it is indeed an oddity in the texts that gets Subsurface confused and not something in the structure. Otherwise more people would have run into this.

dirkhh, 2018-09-05:
Any update on this?

bstoeger, 2018-09-05:
Are there many pictures in the log-book? Could be the conversion of thumbnails to the new format (though it should show a dialog).

If you don't want to share the log-book (in private), could you run it under `gdb` and provide a backtrace:
1) `gdb ./subsurface`
2) CTRL-C
3) `where`
4) possibly switch thread context with `thread n` where n=1, 2, .. and `where` again.

dirkhh, 2018-09-13:
no update in > 1 week

closed 2018-09-13 by dirkhh