subsurface/subsurface #1628: E: Version '4.8.1.1-xenial' for 'subsurface' was not found

phillefjonk, 2018-09-02:
I following the instructions here: [https://subsurface-divelog.org/download/](url)
to install Subsurface on my Lenovo X1 laptop, running Linux Mint 19

I run the following commands in a terminal

sudo apt-add-repository ppa:subsurface/subsurface
sudo apt-get update
sudo apt-get install subsurface=4.8.1-1~xenial

I get the following error message when doing:  $ sudo apt-get install subsurface=4.8.1.1-xenial

E: Version '4.8.1.1-xenial' for 'subsurface' was not found 

My system info:

System:    Host: philip-ThinkPad-X1-Carbon-6th Kernel: 4.15.0-33-generic x86_64 bits: 64 gcc: 7.3.0
                 Desktop: Cinnamon 3.8.8 (Gtk 3.22.30-1ubuntu1) dm: lightdm Distro: Linux Mint 19 Tara
Machine:   Device: laptop System: LENOVO product: 20KH007BMX v: ThinkPad X1 Carbon 6th
                  UEFI: LENOVO v: N23ET40W (1.15 ) date: 04/13/2018

mturkia, 2018-09-02:
Unfortunately we are lacking version 4.8.1, so you will need to grab 4.8.0 instead.

phillefjonk, 2018-09-02:
Thanks mturkia

However
$ sudo apt-get install subsurface=4.8.0
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Version '4.8.0' for 'subsurface' was not found

Which syntax of the version number should I use?

phillefjonk, 2018-09-02:
PS  I use the Suunto Eon core dive computer DS

mturkia, 2018-09-02:
Should be 4.8.0-1~xenial. I was on mobile phone, so didn't even try to type
it all in (especially since I couldn't verify the exact version string
needed)

phillefjonk, 2018-09-02:
Still no luck
$ sudo apt-get install subsurface=4.8.0-1~xenial
[sudo] password for philip: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Version '4.8.0-1~xenial' for 'subsurface' was not found

Is ppa:subsurface/subsurface the right repo?

dirkhh, 2018-09-05:
This should be fixed. We now have 4.8.1 for xenial and bionic

closed 2018-09-05 by dirkhh