subsurface/subsurface #1634: Large UDDF file crashes Subsurface

bluppfisk, 2018-09-03:
### Describe the issue:
- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Subsurface crashes upon importing a large UDDF file (13 MB; 273 dives; 300,000+ lines). Probably runs out of memory.

Segfault in RulerNodeItem.recalculate

### Operating system:
* Ubuntu 18.04 on a Dell XPS 13
* MacOS X 10.13.6 on Macbook Pro 2014
no other OSs were tested

### Subsurface version:
latest stable 4.8.0-2
no other versions are tested

### Steps to reproduce:
Import large UDDF file ([sample from Deepblu](https://drive.google.com/open?id=0B2JazhCkY1_7R051Z2hPUUg0Q2NIV1JBaWtmTEVzN3I3V25Z))

### Current behavior:
Subsurface becomes unresponsive and finally crashes (8 GB RAM system)

### Expected behavior:
Subsurface handles large files in chunks to avoid crashing

### Additional information:

### Mentions:

janmulder, 2018-09-03:
Did a little investigation in this issue. The following 1 dive excerpt (dive nr 5 from the original data) shows the exact same crash: 

[stripped.uddf.txt](https://github.com/Subsurface-divelog/subsurface/files/2345358/stripped.uddf.txt)

This results in the conclusion that the title of this issue is not correct. Further notice that (for example) the `divetime` attribute in the suspect dive is strange. Values like 1535853360. Notice that the UDDF spec says: "The absolutely passed time since the beginning of the dive, given in seconds as a real number.". 1) it is not a real number, and 2) the value 1535853360 seconds is a dive longer than I have ever made :-) 

The question now is: how robust needs Subsurface be on wrong UDDF/XML input? A crash is obvious not nice, but a full blown UDDF/XML validator is not the purpose of Subsurface.

Notice that I did not investigate if the weird `divetime` data is the reason for the crash.

bluppfisk, 2018-09-03:
True, this UDDF file _does_ contain tests with diving hardware in pressurised chambers etc. However, I do not think Subsurface should crash at this point. If it cannot deal with a number that long, at least I think it  should skip over dives with problematic data and warn that x number of dives have not been imported.

bluppfisk, 2018-09-03:
further investigation shows that is indeed the divetime value that is problematic (it requires 31 bits). Further removing samples didn't solve the issue until I indeed changed the divetime value to a reasonable number:

[stripped.uddf.txt](https://github.com/Subsurface-divelog/subsurface/files/2345500/stripped.uddf.txt)

I can't find the bit of code in the repo where UDDF is parsed but the data type could potentially be changed and/or the value tested for compliance.

I will update the title according to your findings.

janmulder, 2018-09-03:
Some more analysis. The bogus UDDF data is parsed correctly, and finally ends up to be plotted. A tiny part of the plot logic fails hard on the huge dive time and crashes. With a trivial 1 line of code, I can prevent **this** crash. 

But, next time somebody invents a different bogus UDDF input, etc. We are not an UDDF validator. I definitely do not like crashes, but "testing for compliance" in a realistic way does not seem possible as the UDDF spec seems not the define maximum values.

And @bluppfisk: just delete all dives newer than July 14, 2017, and the rest of the data will import just fine.

bluppfisk, 2018-09-04:
fair enough. And I'll leave whether insane data is allowed to crash subsurface up to the developers.

closed 2018-09-04 by bluppfisk