subsurface/subsurface #1639: Feature request: capability to add qml files to App, without rebuilding all

janiversen, 2018-09-05:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [x ] New feature request
- [ ] Discussion request

### Issue long description:

When working with QML, it is kind of try and fail. Currently the process is:
- edit qml
- build Subsurface-mobile.app (which takes time, due to make checking everything)
- run

It should be possible to do something like:
- edit qml
- run script xyz (or make xyz) that just updates the qml in the app directory

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2018-09-05:
Since we use Qt's resource system to build the QML into the app executable, I don't think this is possible

janiversen, 2018-09-05:
?? sounds illogical that we cannot bypass the c++ compile steps, link corelib, add languages....and only have the resouce generation and final linking. But of course you might be right I opened this issue to see if there were interest in such a possibility

janiversen, 2018-09-05:
I will see what I can do local.

closed 2018-09-05 by janiversen

janmulder, 2018-09-05:
@janiversen: are you talking about building the app for device? or building mobile-on-desktop version? The last is pretty fast and in a lot of cases sufficient for testing.

janiversen, 2018-09-05:
@janmulder I actually talk about both. I use mobile for quick testing, and device for final testing. 

“pretty quick” is relative, if I just change a single qml file, it takes 3-4 to generate the executable, and it e.g. everytime complies some c++ files, as well as copy a lot of language files.

janmulder, 2018-09-06:
Yes, “pretty quick” in my case (mobile-on-desktop). Just tested a 1 letter useless QML change. Compile is ready in 3 seconds, and does not mess with anything other than the QML change. Obviously, I'm using Qtcreator and not some build script from the repo, and running on Linux. You are probably doing something else on Mac (and not sure what your 3-4 is, seconds or minutes).

janiversen, 2018-09-06:
it is 3-4 minutes, but you are right I use the normal build script...using qtCreator is a good idea....I gave up on creator because I really wanted to use the designer, but it do not work with Kirigami, or fo you also have a solution to that?

janmulder, 2018-09-06:
Never even looked at the Qt designer ... so no solution there.