subsurface/subsurface #1642: Default file name not backwards compatible

bstoeger, 2018-09-06:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

It just happened to me that an older and a recent version open different default logs.
Looking into the config file I see both versions of the file:

new:
```
[General]
CloudStoragesubsurface_webservice_uid=XXX
GeneralSettingsdefault_filename=XXX
loadFromCloudXXXXXt=true
subsurface_webservice_uid=XXX
```
Note line 4: The email is in the key - is that on purpose? Looks ugly.

old:
```
[GeneralSettings]
auto_recalculate_thumbnails=true
default_cylinder=AL80
default_file_behavior=1
default_filename=XXX
default_git_branch=master
defaultsetpoint=1100
extract_video_thumbnails=true
extract_video_thumbnails_position=84
ffmpeg_executable=ffmpeg
o2consumption=720
pscr_ratio=100
use_default_file=true
```

I'm marking this as a bug, even though I'm not 100% certain it should be considered as such. It would be borderline OK if old->new settings are taken but new->old breaks. Is that the case.

In any case, I'm not sure that I like the split between `GeneralSettings` and `General` as well as the gratuitous change of the config file format.

### Operating system:

Kubuntu 18.04

### Subsurface version:

current master + local patches.

### Steps to reproduce:

Start old and new versions.
### Current behavior:

Different config-file layouts.
### Expected behavior:

Same config-file layouts.
### Additional information:

### Mentions:

@janiversen

janiversen, 2018-09-06:
seems like qPref uses “General” instead of “GeneralSettings”, I have to check the old code and correct if there are a difference...however I miss many entries in what you list as “new”. that is kind of strange.

The names within the section should be identical to the old ones, but let me double check.

loadFromCloudXXX = 
(without the last t I hope) is indeed ugly, but that part was in the old code and I just standardized it.
I have the intention of rewriting thst part of the code.

bstoeger, 2018-09-06:
Oh - this was a `QMLManager` variable and became a system wide preferences thing, right..?

janiversen, 2018-09-06:
It was a variable somewhere outside qPref and SettingsObjectWrapper, but without analyzing I cannot say where.

I see the analysis is already in good hands, so no need for me to analyze.
thanks.

dirkhh, 2018-09-06:
We do concatenate strings in the Settings at times (because those aren't really supposed to be human readable). But what all this indicates to me is that I was way to negligent when dealing with all those changes and should have been far more careful to ensure that things don't change on disk.

Given that I want to do a 4.8.2 release I guess I will have to dig into that morass, whether I want to or not.

janiversen, 2018-09-06:
Not only you, these are 2 embarrasing errors that should not have happened....good thing are that they are isolated in 2 individual commits, the loadfromcloud can be reverted without problems, and I recommend correcting the group variable for the name problem.

Of course to me the better alternative to is solving the problem, which I will do, whether you want it as a pr or not seems to be a different matter.

janiversen, 2018-09-06:
Just checked, SettingsObjectWrapper used group "GeneralSettings" for all default values, including default filename.

I cannot see where qPref* writes "General" it seems all to be "GeneralSettings".

However I had a missing / in the default_filename. 

Sending PR with revert of loadfromcloud (to avoid discussions) and a commit to add the slash, and will watch how it is received.

janmulder, 2018-09-07:
@janiversen: and what about the `CloudStoragesubsurface_webservice_uid` that seems to missing / as well.

janmulder, 2018-09-07:
See #1642

dirkhh, 2018-09-07:
> @dirkhh made a lot of changes, which I suppose will secure 4.8.0 and master write the same properties. I do not know if his work is finished.

And by that you mean "Dirk added one commit that stopped us from dumping all the default values into the settings"

janiversen, 2018-09-07:
no I mean exactly I wrote, that I do not know if your work is finished. I thought you wanted 4.8.0 and master to be identical on disk, which to me is more than just the default part.

Is it really nessecary to turn every sentence upside down....I know you are irritated, but please remember you were the one asking me not to leave. I am sorry that I responded to something that clearly was another error in my code.

dirkhh, 2018-09-13:
I think this is fixed now, @bstoeger ?

closed 2018-09-13 by dirkhh