subsurface/subsurface #1648: Weird/buggy webservice UUID behavior (part 1)

janmulder, 2018-09-07:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Some time ago, I forgot my dive computer, so my buddy (also Subsurface user, including using GPS logging using mobile) exported the dive from his Subsurface logbook to complete my logbook. Just verified that his XML export contained his webservice UUID. Not immediately sure this is a bug, but it is al least a privacy leak, as I can now see the logged locations for that webservice UUID. Interestingly, this webservice UUID was the only setting that is exported (and I do not see any reason why we might want to export this). It was likely exported with official version 4.7.8. And just tested with the current master: export does export a webservice UUID. To me, this is a bug. Let call it bug 1).

But it is getting better. As I'm using a local git repo for storage of my logbook, I can see that the imported webservice UUID from the buddy ends up in the persistent and shared preferences (the 00-Subsurface file in the git storage). This import was back in May 2018, using Subsurface:4.7.8.82. Up to now, I've not found a way to reset this value to my own webservice UUID (other then commit a change to my git storage). And obviously, as it are persistent and shared preferences that buddies UUID ends up on my mobile devices as well (verified looking in the git repo from the cloud storage on a rooted Android device, and when syncing GPS data in the app, it shows in the log data as well). One part I do not understand here, is, why is my own webservice UUID never stored in the persistent preferences, but an imported one is? This confuses me big time, and lets call this bug 2)

And yes, we should get rid of that webservice UUID altogether (at lease visible to the user), but that is plan B.

dirkhh, 2018-09-07:
(1) privacy: agreed. I want that whole web service to go away and simply apply the GPS fixes "on device". I haven't taken the plunge to actually implement that (or "unimplement that"), but I think that's the sane direction. I haven't pushed my GPS fixes to the service in ages.
The reason the XML file contains that is by now bogus, as you can now look up your ID from your Cloud Storage account. But in older versions that wasn't the case and there this saved you from having to type it in on every computer or VM that you used.

(2) that's strange; your own user ID (which you enter in the preferences) should then overwrite your buddies. This sounds like a much more serious bug to me than the privacy issue in (1)

We definitely want to fix both issues for 4.8.2 - and we should discuss on the mailing list the idea to just switch to not using the GPS webservice anymore

janmulder, 2018-09-07:
See also issue #1013 (yes, an old one, but related).

janmulder, 2018-09-08:
Ok, I do understand (2). To save an updated webservice UUID into the persistent and shared preferences (in other words: in your logbook!), you need to save the logbook (yes, obvious in hindsight). Simply editing it in the preferences screen, hit save on that screen, does save it in your local .conf file (assuming save_userid_local=true), but this does not save your logbook. The same is true for all possible settings that go into the logbook to be shared between devices (like toggles for the profile display). 

Furthermore, things gets more and more confusing when same settings are stored in the logbook and in the .conf data (especially when these values differ). For example, the UUID from the logbook takes preference over the .conf data.

There is a (simple) workaround for issue (1) as well. Just set "Save ID locally" to false before export. This prevents the export of the webservice UUID.

This all said ... yes, its all very confusing and convoluted, but simply calling it bugs like I did yesterday is also not right. It is worse: its poor design and obviously undocumented.

So I propose close of this one, and this leaves the question "Do and can we abandon the webservice completely?".

dirkhh, 2018-09-08:
see email to mailing list

janmulder, 2018-09-10:
Close this as the decision is made to remove the GPS webservice.

closed 2018-09-10 by janmulder