subsurface/subsurface #1654: travis mac build problems.

janiversen, 2018-09-09:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

The Travis mac build have started to generate a ton of warnings from libdivecomputer, all like:

ld: warning: object file (/Users/travis/build/Subsurface-divelog/install-root/lib/libdivecomputer.a(version.o)) was built for newer OSX version (10.12) than being linked (10.11)

It seems there is a make problem with libdivecomputer. This is potential dangerous, it we use or start using a API call that are different between 10.12 and 10.11, API are backwards compatible (compile for 10.11 but link for 10.12) but not forward compatible.

### Operating system:

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2018-09-09:
Interesting. In `.travis.yml` we request `xcode8` for our Mac build and that is supposed to be macOS 10.11. I've seen these warnings on and off before (and they are annoying, but not really an issue, unless they happen in **my** builds for release binaries), but I really wonder how they can happen on a macOS 10.11 VM.

dirkhh, 2018-09-09:
Interestingly, it looks like the Travis build VMs run 10.11 but don't have a 10.11 SDK installed, so we explicitly end up setting `-mmacosx-version-min=10.12` when building things.
Again, annoying, but not a problem.

PR #1671 (2018-09-13): Mac: hard code macosx-version-min as 10.11, regardless of available SDKs
This should never be wrong, and should avoid the warnings on Travis.

Fixes #1654

### Describe the pull request:

- [ ] Bug fix
- [ ] Functional change
- [ ] New feature
- [ ] Code cleanup
- [x] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

### Changes made:

### Related issues:

### Additional information:

### Release note:

### Documentation change:
commit d94f8136 (2018-09-13): Mac: hard code macosx-version-min as 10.11, regardless of available SDKs

This should never be wrong, and should avoid the warnings on Travis.

Fixes #1654
closed 2018-09-13 by dirkhh