subsurface/subsurface #1679: Problem exporting dive data when fields contain quotes (" ")

p-neves, 2018-09-14:
### Describe the issue:

- [X] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When exporting dive data as *.csv file, fields containing quotes (" ") are broken into other fields. e.g if a Dive location name contains " ", this field is broken before the quotes and the remaining characters are inserted in the next field (gps)...

### Operating system:

Arch Linux

### Subsurface version:

Latest master

422-gd94f81362396

### Steps to reproduce:

1) Select some dives containing quotes on any field (" ") or 
2) Select File -> Export -> Dive details as csv
3) Open exported file and inspect csv file 

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

bstoeger, 2018-09-17:
If anybody wants to tackle this, the XSLT-file in question is
```
xslt/xml2manualcsv.xslt
```
There are some discussions on the net on how to CSV-quote using XSLT. Unfortunately, it's all Chinese to me. Up to now I was completely oblivious of XSLT's existence.

mturkia, 2018-09-17:
I just implemented quoting of " in the exported CSV. This is done with double quote (""). However, when trying to import such an export (or anything else) I get nothing. It seems that the CSV import is currently broken, also in the release. Or might it be that all my CSV files are corrupt? (I don't have any of my old files currently available so testing with manually modified export.)

bstoeger, 2018-09-18:
> Or might it be that all my CSV files are corrupt?

In my case, the CSV contains unbalanced HTML. If this turns out to be the problem, I could turn my quick&dirty XMP-Parser into an HTML-fixer... :-P

commit c44ece0e (2018-10-07): CSV export: quote quotes

If a text field contains quotation mark ("), encode this with double
quote ("").

Fixes #1679
closed 2018-10-07 by dirkhh