subsurface/subsurface #1699: AMV graph in desktop version wrong when using more than 1 bottle

kleinerChemiker, 2018-09-18:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

I am sidemount diving with two bottles. One line in the AMV graph seems OK. It consists of downward parts and horizontal parts. The other graph is an downward line without any change. Also the colors are on both graphs exactly the same.

Ich tauche sidemount mit 2 Flaschen, daher 2 AMV Graphen. Ein Graph ist normal, Teile zeigen sinkenden Druck an, Teile des Graphen sind waagerecht, wenn ich gerade aus der anderen Flasche atme. Der andere Graph ist jedoch eine gerade Linie mit sinkenden Druck. So als wenn nur Anfang und Enddruck durch eine Gerade verbunden wäre. Ebenso sind die Farben auf beiden Graphen ident. 

### Operating system:

Windows 10

### Subsurface version:
4.8.1

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

[image]

bstoeger, 2018-09-19:
Hi @kleinerChemiker, could you perhaps post an export (Subsurface XML) of the dive? Note that in the current release we have an "anonymize" function, which removes most textual data (but not GPS coordinates).

kleinerChemiker, 2018-09-22:
It seems the data is missing in the file. The question is, is the data also missing in the diving computer or is it lost when transfered from the dive computer to Logtrack. Unfortunately I cannot read the diving computers log since it is in some binary format.

[dive.zip](https://github.com/Subsurface-divelog/subsurface/files/2407787/dive.zip)

kleinerChemiker, 2018-10-04:
Usually I use the android subsurface app to download the dives and sync with the cloud. Of course I could use Logtrack to test the download. I could even download the whole userfile from the G2 by USB. However I would share this file only by mail with a dev and not post it here since I dont know which data is stored in the file.