subsurface/subsurface #1702: Unmet dependencies on Ubuntu 18.04.1 LTS - libcurl3

maxdraki, 2018-09-19:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Cannot install Subsurface 4.8.2 (and recent versions) on Ubuntu Ubuntu 18.04.1 LTS due to unmet dependencies. If I try to force libcurl3 to install I break curl and playonlinux (that I use for various wine apps)

```
The following packages will be REMOVED
  curl libcurl4 playonlinux
The following NEW packages will be installed
  libcurl3
```
### Operating system:
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.1 LTS
Release:	        18.04
Codename:	bionic

### Subsurface version:
This occurs with 4.8.2, but also earlier versions - I have been using AppImage, but would prefer to actually install it

### Steps to reproduce:
Add ppa
sudo apt install subsurface

### Current behaviour:
Installation fails with:
```
The following packages have unmet dependencies.
 subsurface : Depends: libcurl3 (>= 7.18.2) but it is not going to be installed
```

### Expected behavior:
Installation

### Additional information:
N/A

### Mentions:

mturkia, 2018-09-19:
Where are you installing the package from? What does following show:
apt-cache policy subsurface

When looking at the dependencies on my Ubuntu, Subsurface depends on libcurl4 which is correct. Besides, the dependencies are validated on Launchpad during the creation of the package.

maxdraki, 2018-09-27:
Sorry for the late reply - I've been diving in the Red Sea :)

- and thanks for the suggestion to check the package policy - I still had the distribution set to 'xenial' - set it to 'bionic' and all is good..

my mistake - but thanks for the help

closed 2018-09-27 by maxdraki

danbowkley, 2018-09-30:
Stupid Linux question...how do I change that? Mine says Artful, I'm also on Bionic and my Subsurface crashes immediately on loading.

maxdraki, 2018-09-30:
launch "software and updates", go to "other software" and then edit the Subsurface entry - update the distribution as per the screenshot to 'bionic'

[image]