subsurface/subsurface #1715: Desktop: BT/BLE connection is not restored from saved DCs

janmulder, 2018-09-24:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
Like #1714, I see a similar issue on the desktop, but as the root cause is very likely to be different, I open a new issue for this. Starting the application, the download from DC is correctly filled from the settings. That is, it shows the last used DC (and as I have only one DC in use, that is just fine). However, after the first download a shortcut button is added. Hitting this, clears my BT connection string.

Inspecting the code, this seems obvious. 1) the connection pull down only contains the devices that are found on the computer (in my case /dev/ttyS[0-3]). Unlike mobile, there are no BT addresses added to this list. 2) as the value from the shortcut button (correctly coming from the settings) is simply not in this pull down, a findText() in this list of options from the pull down menu is always going to fail. The only way to get the BT address back is doing a new scan in the "get remote BT device) sub-window, which defies the logic behind this new functionality.

So, I'm confused. Has @dirkhh as the main developer of this new functionality ever seen this working?

### Operating system:
 What OS are you running, including OS version and the language that you are running in: up to date Arch Linux

### Subsurface version:
What version of Subsurface are you running? --> master
Does the same happen on another Subsurface version? --> na., first version in master
Are you using official release, test build, or compiled yourself? --> self
Provide Git hash if your are building Subsurface yourself. --> 9071dd42f2f37

### Steps to reproduce:
Simply hit a BT stored DCs button, and see that it does not populate the connection.

dirkhh, 2018-09-24:
Yes, I have - because I wrote this in a VM (I intentionally develop on many different machines, some Macs, some Linux) and so it was only tested with serial ports.
The bug report shows how dumb that was, but the fix seems rather obvious - if we don't find it, add it to the model.

janmulder, 2018-09-25:
Merged PR does the job with a small consistency remark in #1716, but no reason to keep this issue open. Closed.

closed 2018-09-25 by janmulder