subsurface/subsurface #1740: Subsurface-4.8.3-x86_64.AppImage segfaults at startup

cluon-mike, 2018-09-30:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

$ ./Subsurface-4.8.3-x86_64.AppImage
Auto configuration failed
139712809015744:error:25066067:DSO support routines:DLFCN_LOAD:could not load the shared library:dso_dlfcn.c:185:filename(libssl_conf.so): libssl_conf.so: cannot open shared object file: No such file or directory
139712809015744:error:25070067:DSO support routines:DSO_load:could not load the shared library:dso_lib.c:244:
139712809015744:error:0E07506E:configuration file routines:MODULE_LOAD_DSO:error loading dso:conf_mod.c:285:module=ssl_conf, path=ssl_conf
139712809015744:error:0E076071:configuration file routines:MODULE_RUN:unknown module name:conf_mod.c:222:module=ssl_conf
QMutex: destroying locked mutex
Segmentation fault

### Operating system:

$ uname -a
Linux ... 4.18.0-1-amd64 #1 SMP Debian 4.18.8-1 (2018-09-18) x86_64 GNU/Linux
$ cat /etc/debian_version 
buster/sid

### Subsurface version:

4.8.3 release downloaded from website

### Steps to reproduce:

execute appimage binary

### Current behavior:

segfaults on start

### Expected behavior:

### Additional information:

### Mentions:

micah, 2019-03-03:
I have this same problem. I'm able to keep it from segfaulting immediately if I select another dive right when it starts up... but then if I try and use the map, it will segfault pretty quickly.

This happens on 4.8.5 AppImage as well.

neolit123, 2019-03-03:
i cannot reproduce this problem with the 4.8.5 AppImage on:
```
Distributor ID:	Ubuntu
Description:	Ubuntu 17.10
Release:	17.10
Codename:	artful
```

micah, 2019-03-03:
I did this:

```
$ sudo ln -s /usr/lib/x86_64-linux-gnu/libssl.so.1.1 /usr/lib/x86_64-linux-gnu/libssl_conf.so
```

and things worked again. This is obviously not a good solution and instead there is likely a bug in Debian's libssl package.

hhassey, 2019-05-29:
Can Confirm same crash behavior with Debian Buster and App image 4.8.6

hhassey, 2019-05-31:
@micah Yes Thanks apparently it works fine now. I have not downloaded my last few dives but it seems ok.

glance-, 2019-11-10:
This is also worked around with #2296

closed 2019-11-10 by glance-