subsurface/subsurface #1761: serial_ftdi_read() read timeout potentially too long

bperrybap, 2018-10-07:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

The function serial_ftdi_read() in core/serial_ftdi.c is not handling a read timesout properly.
This can cause the timeout  period while waiting for read data to be 10x or even 100x longer than it should be when there are read issues on the data cable particularly when using Android and USB OTG cables.
i.e. a  5 second read timeout for not receiving data can be as long as 7 minutes.

When things are working normally without communication errors, this bug does not create any problems for downloading from the dive computer; However, when there are communication issues, an extended timeout can cause libdivecomputer retries to fail as the data cable and/or divecomputer can timeout long before libdivecomputer can re-send the command to the dive computer to retry the command.

The issue is that the code has a loop that checks for characters every 1ms.
It loops 1 time for each 1ms in the desired timeout. The issue is that the the code inside the loop is not accounting for the amount of time consumed by the code inside the loop so the actual timeout period can be much longer than the desired timeout period.
The main culprit for the added time is the ftdi_read_data() call. This time used by the function can vary substantially and can be as long as 140ms when using libusb on Android with a USB OTG adapter and FTDI data cable.  - (as is used when using a FTDI data cable on Pelagic dive computers)
The solution is to still sleep for 1ms between polls for characters but not blindly loop for the number of milliseconds in the timeout period but rather break out of the loop when the elapsed time in the loop exceeds the desired timeout period.

### Operating system:

Pretty much all but I noticed issues when using Android 4.4.4 and Android 8.0.0

Galaxy S4 (Android 4.4.4) Galaxy S7 (Android 8.0.0)
Atmos Aeris AI using Oceanic 04.9600 USB data cable

### Subsurface version:

Pretty much any subsurface version, but the extra long timeouts are most noticeable on the older versions of subsurface mobile on Android from late summer of 2017.

### Steps to reproduce:

It can be difficult to fully reproduce the super extra long timeouts as the excess time used by ftdi_read_data() can vary substantially depending on the version of Android and the version of Subsurface mobile.
To reproduce / test simply connect the FTDI data cable to the phone but not to a Dive computer and try to do a download from an Aeris Atmos AI dive computer.
The Subsurface libdivecomputer code has read timeouts and does retries and the timeouts can be much longer than expected - but it depends on the version of Subsurface mobile.
The earlier Subsurface versions from the summer of 2017 were much better at demonstrating the issue.

### Current behavior:

Latest versions of Subsurface Mobile do not work on Galaxy S4 with Android 4.4.4
It can't get permission to use the USB device as the USB device dialog does not come up anymore when the USB cable is plugged in on the past few Subsurface mobile builds from the Play Store.

Subsurface Mobile Version 2.1.3(4.8.2.5) on a Galaxy S7 seems to timeout without an excessively long timeout period when the dive computer is not attached. However, even though the existing code is not currently creating/triggering an excessive timeout, the issue/bug still exists.

### Expected behavior:

serial_ftdi_read() returns after the specified timeout period with a TIMEOUT status if all the data wasn't read.

### Additional information:

I had a solution that worked on Android that I was going to commit and generate a pull request to fix it, but it depends on gettimeofday() to be able to track elapsed time. For some reason WIN32 does not include this call. I found a substitute for this for WIN32 but I have no idea about the other OSes.
I'm not sure what the final soultion needs to be and whether or not Subsurface is interested in pulling in this WIN32 gettimeofday() function.

Below is the serial_ftdi_read() code I was using:
```
static dc_status_t serial_ftdi_read (void *io, void *data, size_t size, size_t *actual)
{
	ftdi_serial_t *device = io;

	if (device == NULL)
		return DC_STATUS_INVALIDARGS;

	// The total timeout.
	long timeout = device->timeout;

	// Simulate blocking read as 10s timeout
	if (timeout == -1)
		timeout = 10000;

	int backoff = 1;
	unsigned int nbytes = 0;

        struct timeval tvstart, tvnow;
        gettimeofday(&tvstart, 0);

	while (nbytes < size) {
		int n = ftdi_read_data (device->ftdi_ctx, (unsigned char *) data + nbytes, size - nbytes);
		if (n < 0) {
			if (n == LIBUSB_ERROR_INTERRUPTED)
				continue; //Retry.
			ERROR (device->context, "%s", ftdi_get_error_string(device->ftdi_ctx));
			return DC_STATUS_IO; //Error during read call.
		} else if (n == 0) {
                        gettimeofday(&tvnow, 0);
                        // check to see if too much time has elapsed
                        if( ((tvnow.tv_sec - tvstart.tv_sec) * 1000) + ((tvnow.tv_usec - tvstart.tv_usec)/1000) >= timeout) {
				ERROR(device->context, "%s", "FTDI read timed out.");
				return DC_STATUS_TIMEOUT;
			}
			serial_ftdi_sleep (device, backoff);
		}

		nbytes += n;
	}

	INFO (device->context, "Read %d bytes", nbytes);

	if (actual)
		*actual = nbytes;

	return DC_STATUS_SUCCESS;
}
```
Here is a WIN32 gettimeofday() function if that is desired route:
```
// for win32 a gettimeofday() is needed.
// one can be found here: https://stackoverflow.com/questions/1676036/what-should-i-use-to-replace-gettimeofday-on-windows
#ifdef _WIN32
/*
 * gettimeofday.c
 *    Win32 gettimeofday() replacement
 *
 * src/port/gettimeofday.c
 *
 * Copyright (c) 2003 SRA, Inc.
 * Copyright (c) 2003 SKC, Inc.
 *
 * Permission to use, copy, modify, and distribute this software and
 * its documentation for any purpose, without fee, and without a
 * written agreement is hereby granted, provided that the above
 * copyright notice and this paragraph and the following two
 * paragraphs appear in all copies.
 *
 * IN NO EVENT SHALL THE AUTHOR BE LIABLE TO ANY PARTY FOR DIRECT,
 * INDIRECT, SPECIAL, INCIDENTAL, OR CONSEQUENTIAL DAMAGES, INCLUDING
 * LOST PROFITS, ARISING OUT OF THE USE OF THIS SOFTWARE AND ITS
 * DOCUMENTATION, EVEN IF THE UNIVERSITY OF CALIFORNIA HAS BEEN ADVISED
 * OF THE POSSIBILITY OF SUCH DAMAGE.
 *
 * THE AUTHOR SPECIFICALLY DISCLAIMS ANY WARRANTIES, INCLUDING, BUT NOT
 * LIMITED TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR
 * A PARTICULAR PURPOSE.  THE SOFTWARE PROVIDED HEREUNDER IS ON AN "AS
 * IS" BASIS, AND THE AUTHOR HAS NO OBLIGATIONS TO PROVIDE MAINTENANCE,
 * SUPPORT, UPDATES, ENHANCEMENTS, OR MODIFICATIONS.
 */

#include "c.h"

#include <sys/time.h>

/* FILETIME of Jan 1 1970 00:00:00. */
static const unsigned __int64 epoch = ((unsigned __int64) 116444736000000000ULL);

/*
 * timezone information is stored outside the kernel so tzp isn't used anymore.
 *
 * Note: this function is not for Win32 high precision timing purpose. See
 * elapsed_time().
 */
static int
gettimeofday(struct timeval * tp, struct timezone * tzp)
{
    FILETIME    file_time;
    SYSTEMTIME  system_time;
    ULARGE_INTEGER ularge;

    GetSystemTime(&system_time);
    SystemTimeToFileTime(&system_time, &file_time);
    ularge.LowPart = file_time.dwLowDateTime;
    ularge.HighPart = file_time.dwHighDateTime;

    tp->tv_sec = (long) ((ularge.QuadPart - epoch) / 10000000L);
    tp->tv_usec = (long) (system_time.wMilliseconds * 1000);

    return 0;
}

#endif
```

torvalds, 2018-10-07:
Ugh.

If we have to make excuses for Windows anyway, I'd much rather just do it the same way we already do in serial_ftdi_sleep(), and just have a "this is how it works on windows, this is how sane people do it".

So just add a "serial_ftdi_get_msec() that is the "reverse" of the serial_ftdi_sleep() function, and just returns number of msec's since some random time.

I'll send a pull request for a suggested fix. NOTE! I can't test the windows side, but the code looks obvious.

closed 2018-10-07 by torvalds

neolit123, 2018-10-07:
hello @torvalds @bperrybap ,

gettimeofday() is portable to Windows because we are using the MINGW compiler.
that said we can avoid gettimeofday() by using clock() which is a function in the C standard.
e.g. like so:
https://www.geeksforgeeks.org/how-to-measure-time-taken-by-a-program-in-c/

neolit123, 2018-10-07:
ok, on `clock()` - it's quite unreliable, indeed.

GetTickCount() should work as the Windows alternative, but as already mentioned we are not switching to MSVC anytime soon so gettimeofday() was just fine given MINGW already bundles an implementation.

dirkhh, 2018-10-07:
I've merged Linus implementation - but until we have binaries from the CI build, you can get a test binary here if you want to test @bperrybap : https://transfer.sh/XFnU9/subsurface-4.8.3-69-g9e24a8264e89.exe

dirkhh, 2018-10-08:
All our OSs (well, with the exception of iOS) use libusb. I got and thought you were on Windows - but of course now that I look I see that you mention Android.
I assume you can build from source, otherwise there's an APK on our releases page:
https://github.com/Subsurface-divelog/subsurface/releases/tag/continuous
(that isn't signed with the same certificate as our official binaries, so you'll have to uninstall the previous version, first, before you can install this one)

PR #1762 (2018-10-07): Recent fixes

### Describe the pull request:

- [x] Bug fix
- [x] Functional change
- [x] New feature
- [ ] Code cleanup
- [ ] Build system change
- [ ] Documentation change
- [ ] Language translation

### Pull request long description:

Various random misc fixes 

### Changes made:

1) implement the "get_name()" function the BLE iostream, needed for the Aqualung i770R and i300C

2) make the ftdi user-space driver hopefully get the right timeouts across different operating systems

3) fix a build warning

### Related issues:

Fixes #1761 

### Additional information:

### Release note: