subsurface/subsurface #1787: Create filter option for tank size

jkingsman, 2018-10-15:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [x] New feature request
- [x] Discussion request

### Issue long description:
As a diver, I sometimes want to compare my weighting based on a given suit and tank size (e.g. I know I've dived XXX suit with an HP100 before; how much weight did I use?). Adding a filtering option for tank size (along with the already-present tags, person, location, and suit filtering) would solve this problem easily.

One potential risk is that it pushes the boundaries further for "what about a filter for XYZ" to the potential nth degree; not sure how big of a concern that is.

If this is a feature that there is interest in, I'm happy to do my best on a first draft -- I know it would require doc updates, etc. I've been shadowing the mailing list for a while; hopefully this is the right place to post an idea for feedback and discussion -- feel free to delete or let me know a better place to raise this if I've misstepped 🙂

janmulder, 2018-10-15:
As a current workaround, you can add the weight + cylinder to the divelist displayed columns and sort on those. Notice that the divelist only displays the first cylinder (for multi cylinder dives).

> One potential risk is that it pushes the boundaries further for "what about a filter for XYZ" to the potential nth degree; not sure how big of a concern that is.

Well, this is always some sort of concern. There seems room for it in the UI, so in this respect this could be added.

That said, my main concern is that the cylinder type is just a tiny part in the whole buoyancy equation. For example, water type, cylinders with same size/type but different weight, amount of undergarment when dry-suit diving, and even the comfort level of this diver on that dive. So, in this respect: its not useful to me to be to add this to the UI.

jkingsman, 2018-10-15:
That's definitely a fair point re: not actually solving the problem of buoyancy factors; this might not be as helpful as I imagined.

commit 4ec0c650 (2020-10-03): filter: add tank size filter constraint

This is an actual user request.

Fixes #1787.
closed 2020-10-03 by dirkhh