subsurface/subsurface #1800: windows-container build fails when run locally

salvadorcunat, 2018-10-15:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

./scripts/windows-container/travisbuild.sh fails to build grantlee and googlemaps with:

/win/grantlee/templates/scriptabletags/scriptabletags.h:42: Error: Undefined interface

/win/googlemaps/qgeoserviceproviderplugingooglemaps.h:10: Error: Undefined interface

which is weird as the same script works like a charm in travis, and works for @dirkhh too.  So the fail seems related to my system.

### Operating system:

Fails on Debian (buster/sid) currently with:
Qt5-11-1
cmake 3.12.3

Works on travis:
Ubuntu 14.04.5
cmake 3.9.2

Same mxe mingw32 toolchain in both cases.

### Subsurface version:

In current master a6e4de6b5e42

### Steps to reproduce:

from subsurface directory run:
bash ./scripts/windows-container/before_install.sh
bash ./scripts/windows-container/travisbuild.sh

### Current behavior:

For me the build fails for grantlee; if we rerun the travisbuild.sh script, it fails for googlemaps.
 
### Expected behavior:

Build a Subsurface and a smtk2ssrf windows packages and, previously, build their dependencies.
### Additional information:

### Mentions:

dirkhh, 2018-10-16:
I may be completely wrong here, but I wonder if this is an issue of overlap of different builds.
If you check out a fresh Subsurface tree in a subdirectory, does it work?

```
mkdir -p ~/src/test/win
cd ~/src/test/win
git clone git://github.com/subsurface-divelog/subsurface
cd subsurface
bash ./scripts/windows-container/before_install.sh
bash ./scripts/windows-container/travisbuild.sh
```

salvadorcunat, 2018-10-16:
I tried yesterday, in a  brand new dir under /tmp, getting the same results.
Then, manually working on your running docker image, could finally get an smtk2ssrf installer (because grantlee and gogglemaps are not needed in smtk-import).
I'm thinking about some kind of PATHs issue, something travis-ci sets by default and I don't (or viceversa).

dirkhh, 2018-10-16:
But all PATH issues should be inside the Docker image, right? So that's what makes no sense.
The Docker image is creating a logfile (winbuild.log). Could you attach that to this issue?
You'll likely have to compress it to winbuild.log.gz and then attach it as otherwise it won't make it past the silly file extension test of GitHub.

salvadorcunat, 2018-10-16:
jejeje.
See [this thread in stack overflow](https://stackoverflow.com/questions/48995826/which-capabilities-are-needed-for-statx-to-stop-giving-eperm) ,and the [resulting PR](https://github.com/moby/moby/pull/36417#issuecomment-370782801)
travisbuild.sh works fine if --privileged param is added to before_install.sh docker run command.
This should be fixed for docker 18.04 and newer (I'm on 17.05), but weirdly travis-ci is on 17-09.

neolit123, 2018-10-21:
@salvadorcunat 
should we close this issue?

mikeller, 2026-10-17:
Hi @salvadorcunat,

Thanks for tracking down the actual cause yourself, the Windows container build needing --privileged on older Docker versions due to the statx syscall restriction, as you found via the Moby/Docker discussion you linked.

Closing this as you suggested. We'll look at adding a comment to the build script documenting the --privileged workaround for anyone else who hits this on an older Docker version.

closed 2026-10-17 by mikeller