subsurface/subsurface #1821: F-Droid availibility

IzzySoft, 2018-10-24:
Unfortunately, #1429 was closed and set read-only without the question being answered – so please excuse me bringing it up here again:

Thanks for providing the APK with your `releases/` – but those are "debug builds". From a raw look at what's inside (and [what Exodus reports](https://reports.exodus-privacy.eu.org/reports/34080/)), it should be fine with F-Droid. So just asking for a short reply: Is it OK for you to have your app (built and) available there? If so, I'd take care for bringing it there. If not, I'd be disappointed but wouldn't bother you again in the near future :wink:

Thanks!

dirkhh, 2018-10-24:
I'm not interested. But this is an open source project, so go ahead. Just make sure that the packaging makes it clear that people need to contact you and not us if they run into problems.

atdotde, 2018-10-24:
Let me add that it serves nobody if your repository is not recent. We still get bug reports/ complaints about things that were fixed long ago but the user got their copy from their distribution which happened to be many months if not years behind.

IzzySoft, 2018-10-24:
@atdotde F-Droid can fetch updates automatically as soon as you create new tags. My own repo does the same (I indeed just added your debug-build there for now and told my updater to skip the pre-releases, so it should only fetch the "green tags"), so the APK should be quite up-to-date in both (my updater actually checks all 24h; the official repo sometimes takes a little more than a week due to the manual signing part of the process).

@dirkhh thanks! Let me prepare the RFP (Request For Packaging), and then I ping you back to have a look at the description set up to approve. Though I don't understand what people should contact F-Droid about then, as we simply compile your code (I didn't see any "unwanted parts" like proprietary libraries we'd need to patch out – though I didn't check for "blobs" manually). Of course folks should contact us when something's amiss (like a new release that didn't show up, or trouble with downloading from F-Droid).

dirkhh, 2018-10-24:
No, I do NOT want you to just push the builds that we do on Travis to some distribution site. They are unsigned for a reason. They are NEVER intended to be installed by anyone than a developer, using `adb`.
Feel free to use any system you want for the binaries **YOU** make available to users. Also, please make sure that they have a modified name that makes it clear that **YOU** are making them available, not us. Including contact email address, support email address, etc.

dirkhh, 2018-10-24:
"releases" on mobile are independent from desktop releases and while occasionally they happen to be on a tag, they generally don't. Which means that by distributing things just based on the source tree being tagged you may be offering broken packages to your users.

IzzySoft, 2018-10-24:
Oh! So what would you recommend to fetch instead, if not just "the green tags"? And do the specific app releases have a specific tag name maybe – or how can they be identified?

For my repo, I especially chose to skip the "pre-release" tags exactly because of that. And the linked initial issue got the answer "APKs are attached there", without a specific hint to which ones are safe to use.

dirkhh, 2018-10-24:
But that's the whole point. We very carefully do our releases independent of upstream tags.
They list which git commit they are based on in their version string.
If you want to distribute matching releases elsewhere you'll have to redo that same process.
Case in point, our iOS releases often have the same mobile version number, but are not at the exact same commit.

I simply don't want to go through the extra effort of yet another random release outlet. So if you want that alternative repo, you'll need to figure out which version to pick for your users. And you need to build your own packages.

IzzySoft, 2018-10-24:
Thanks for the details, Dirk. That somehow explains why your APKs are debug builds – but I'm still confused why an APK attached to a "green tag" cannot be considered a release/stable.

As I cannot build APKs myself (neither I'm a dev nor do I have a build environment), that would mean I have to remove the APK from my repo again. F-Droid can build; but not having a way to automatically determine what to build from, that would mean manual updates and someone determined to maintain that. I doubt there will be regular updates this way if it involves a "tricky path" to find the matching commit (which you already did, so why not tagging it e.g. `android-v4.3.8`?).

Quite disappointing. Somehow defeats the idea of open source if one is forced to enter a walled garden to get it. How do the APKs come to Playstore then – and what makes it so hard to use the same APK here to download, or mark the commit using a tag so F-Droid can stick to that?

Don't get me wrong: I'm not complaining, I'm just puzzled and try to find a way to make it possible. My devices have no Play Store (or Google Apps); they run LineageOS with microG, and I specifically left the GApps out for good reasons. Using Subsurface on my Linux PC, I'd like to have the matching app on my mobile (who takes the PC along when going for a dive?) – but I'd like to take it from an open environment, without bothering with Google stuff. And I'm not the only one with this :wink:

dirkhh, 2018-10-24:
I think we have different ideas what the purpose of an open source project is.

mikeller, 2026-10-17:
Hi @IzzySoft,

To close the loop on this: dirkhh's position throughout was consistent, the project isn't going to maintain an official F-Droid listing itself, but won't block a third party from packaging and distributing it, provided it's clearly branded as a third-party build with its own support contact, and whoever does it takes responsibility for correctly identifying which builds are genuinely release-quality (since our own CI tags aren't a reliable proxy for that on mobile).

Closing this with that stated position as the answer. Please reopen if the project's stance on this changes.

closed 2026-10-17 by mikeller

IzzySoft, 2026-10-18:
Thanks @mikeller – not sure who would/could pick it up that way. At [IzzyOnDroid](https://izzyondroid.org/), we always ship the APKs built and signed by their developer, so we are out. F-Droid could pick up from tags, build and sign themselves – but I doubt they'd take upon themselves the part of white-labeling and support offer 🤷‍♂️