subsurface/subsurface #1828: New "Connect to Facebook" is not working

ChristianMehrer, 2018-10-26:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
I want to "Connect to Facebook" to be able to share dives on Facebook. This is not working.

### Operating system:
Windows 7

### Subsurface version:
offical release 4.8.3

### Steps to reproduce:
Always when I try to connect to facebook. 

### Current behavior:
Afterwards "Share on Facebook" is still grayed out.

### Expected behavior:
Connecting works without error. Sharing is possible.

### Additional information:
[image]
[image]

seanwhe, 2018-12-24:
Agreed using Ubuntu Bionic and Chromium Browser

Message received is 
Facebook has detected Subsurface isn't using a secure connection to transfer information.
Until Subsurface updates its security settings, you won't be able to use Facebook to log into it.

neolit123, 2019-02-01:
we are  removing the FB feature:
https://github.com/Subsurface-divelog/subsurface/pull/1958

closed 2019-02-01 by neolit123