subsurface/subsurface #1830: OSTC sport: Deco model info in Subsurface differs from setting in computer

hhmeyer, 2018-10-26:
### Describe the issue:

- [ x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:
I have a OSTC sport, now using firmware 10.45. For compatibility reasons with fellow divers the values for GF in my computer are set to 30/85 (default is 30/80). A wrong value is displayed in Subsurface for Mac.

### Operating system:
Android 8.1.0
Mac OS X 10.13.6

### Subsurface version:
Android: Subsurface 2.1.4 (4.8.3.0)
Mac: Subsurface 4.8.3
both official versions

### Steps to reproduce:
1. import a dive to Subsurface mobile on Android
2. sync to the Subsurface cloud service
3. sync Subsurface on Mac with the Subsurface cloud service
4. choose a dive and go to tab Extra info -> Deco model info

### Current behavior:
Wrong data is dispayed: 30/80 (I could not find the value in the Android version.)

### Expected behavior:
The correct data as read from the dive computer should be displayed.

dirkhh, 2018-10-26:
Could you attach a screen shot?

hhmeyer, 2018-10-26:
[image]

Is that sufficiant?

hhmeyer, 2018-10-27:
After of course. If I hadn't changed the value to meet the settings in the other two OSTC in my team (one OSTC plus and one OSTC 2, both with tech software) I wouldn't have cared. This was the first time I changed this settings and I wanted to check.

glance-, 2018-10-29:
The data shown there comes from the dive header. 

Open the dive in the logbook on the DC and take a photo of the info screen, showing the GF there. If that Shows something else than what subsurface shows, please dump the raw dive with dctool and i'll take a look at it.

hhmeyer, 2018-11-06:
Short story: user error
Long version: I set the GF high to 85%, but this setting does not come into effect when the setting for conservatism is still 0 (Zero). Conservatism has to be set to GF to make the manual GF settings active.

Thanks for trying to help.

closed 2018-11-06 by hhmeyer