subsurface/subsurface #1831: Bluetooth dc download from laptop not working.

cyberdiamond, 2018-10-27:
### Describe the issue:

- [x] Bug

### Issue long description:

Bluetooth dc download from laptop now not working.
Toshiba W50 laptop inbuilt Bluetooth now not picked up but used to work fine.

### Operating system:

Windows 10 version 1809 build 17763.104

### Subsurface version:

I noticed the new download box in some of the latest builds
Neither subsurface-4.8.3-263-ge0e3b6360100 or subsurface-4.8.3-271-g3f5f851fcb2d work.

### Steps to reproduce:

Menu > Import > Import from dive computer
Check the Choose Bluetooth import download mode

### Current behavior:

Down the bottom of the Bluetooth download window shows:
"The local Bluetooth adaptor cannot be accessed"

On the right hand side it shows:
Local Bluetooth device details
Name: Not available
Address: Not available

### Expected behavior:

It should pickup/use the inbuilt Bluetooth radio.

### Additional information:

I can still download over Bluetooth with multiple computer if I create Bluetooth COM ports and use the old serial com port over Bluetooth mode.

### Mentions:

cyberdiamond, 2018-10-27:
SW Petrel 2 EXT, Ratio iX3m reb & HW OSTC3+

Yes Bluetooth enabled and working normally and downloads are working over the old serial COM port over Bluetooth method.

Device is paired in control panel.

Scan button is grey'd out see screenshot below.

[image]

neolit123, 2018-10-28:
we seem to be using something that is not available on Windows:
http://doc.qt.io/qt-5/qbluetoothlocaldevice.html
> On iOS and Windows, this class cannot be used because the platform does not expose any data or API which may provide information on the local Bluetooth device.

please try:
http://subsurface-divelog.org/downloads/test/subsurface-4.8.3-17-g53341c037d5a.exe

installers newer than that don't work for me either - scan button disabled.

cyberdiamond, 2018-10-28:
subsurface-4.8.3-17-g53341c037d5a.exe works properly.

[image]

neolit123, 2018-10-28:
@dirkhh 
after building latest Subsurface master, i've discovered that the scan button and control are enabled for me with my custom Qt version.
this on the other hand does not happen for our latest CI built Windows installers.

i cannot seem to figure out what the cause for that may be.

dirkhh, 2018-10-28:
The installers from the windows-container build should be using your custom Qt version. Or did I manage to break that? Crap.

cyberdiamond, 2019-06-24:
This is fixed in the latest builds

closed 2019-06-24 by cyberdiamond

dirkhh, 2019-06-24:
Thanks for the update. MUCH appreciated