subsurface/subsurface #1839: Can't build on

bstoeger, 2018-11-01:
### Describe the issue:

- [ ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request
- [x] Build problem

### Issue long description:

After upgrading to Kubuntu 10.10, on application start (desktop) I only get:
```
./subsurface: relocation error: /usr/lib/x86_64-linux-gnu/qt5/plugins/geoservices/libqtgeoservices_googlemaps.so: symbol _ZN11QGeoMapTypeC1ENS_8MapStyleERK7QStringS3_bbi10QByteArray version Qt_5_PRIVATE_API not defined in file libQt5Location.so.5 with link time reference
```
followed by program termination. :(

I tried erasing the `build` and `install-root` directories to no avail.

### Operating system:

Kubuntu 10.10.
### Subsurface version:

Master.
### Steps to reproduce:

1) Erase the `build` and `install-root` directories
2) Run `build.sh`
### Current behavior:

Program terminates complaining about something related to qtlocation.
### Expected behavior:

Program runs.
### Additional information:

### Mentions:

@dirkhh, @neolit123: help?

bstoeger, 2018-11-01:
Having no map would be OK, but even `./subsurface/scripts/build.sh -build-with-webkit -skip-googlemaps` produces a broken binary. :(

neolit123, 2018-11-01:
i've seen this in cases of wrong private headers.

bstoeger, 2018-11-01:
Can I install these myself or has this to be fixed by [K]Ubuntu?
Sorry - I have no insight at all into this location thing.

neolit123, 2018-11-01:
we pull them in the build.sh script:
https://github.com/Subsurface-divelog/subsurface/blob/master/scripts/build.sh#L149

you might have to debug what already exists or is downloaded by the script in the QT_HEADERS_PATH location.

bstoeger, 2018-11-01:
It seems to fetch the correct headers:
```
Missing private Qt headers for 5.11.1; downloading them...
Cloning into './qtlocation_git'...
Note: checking out 'f8a0ae9109c2ee56a0480cbe62f07239afb31392'.
```
and the headers are found nowhere else on the system.

Weirdly, the symbol it complains about *is* in the file:

```
$ objdump -T /usr/lib/x86_64-linux-gnu/qt5/plugins/geoservices/libqtgeoservices_googlemaps.so | grep _ZN11QGeoMapTypeC1ENS_8MapStyleERK7QStringS3_bbi10QByteArray
0000000000000000      DF *UND*  0000000000000000  Qt_5_PRIVATE_API _ZN11QGeoMapTypeC1ENS_8MapStyleERK7QStringS3_bbi10QByteArray
```

So what does that "link time reference" mean?

neolit123, 2018-11-01:
you can ldd the google maps .so and see what libQt5Location.so.5 it's linking against.
my understand of the relocation error is that it can happen if the dynamic linker (https://linux.die.net/man/8/ld.so) tries to use a wrong .so.
possible cause could be more than one libQt5Location.so.5.

mturkia, 2018-11-02:
Is it Kubuntu 10.10 or 18.10? The first one is ancient. If 18.10, it would have been a real surprise, if the private headers were packaged again (assuming you are using the latest Kubuntu).

bstoeger, 2018-11-02:
> Is it Kubuntu 10.10 or 18.10?

18.10 of course - it's my dyslexia breaking through. Geez.

Things are getting clearer:
```
$ objdump -T -C /usr/lib/x86_64-linux-gnu/qt5/plugins/geoservices/libqtgeoservices_googlemaps.so | grep QGeoMapType
0000000000000000      DF *UND*  0000000000000000  Qt_5_PRIVATE_API QGeoMapType::QGeoMapType(QGeoMapType::MapStyle, QString const&, QString const&, bool, bool, int, QByteArray)

$ objdump -T -C /usr/lib/x86_64-linux-gnu/libQt5Location.so.5 | grep QGeoMapType
000000000009ba40 g    DF .text  00000000000000d6  Qt_5_PRIVATE_API QGeoMapType::QGeoMapType(QGeoMapType::MapStyle, QString const&, QString const&, bool, bool, int, QByteArray const&, QGeoCameraCapabilities const&, QMap<QString, QVariant> const&)
000000000009ba40 g    DF .text  00000000000000d6  Qt_5_PRIVATE_API QGeoMapType::QGeoMapType(QGeoMapType::MapStyle, QString const&, QString const&, bool, bool, int, QByteArray const&, QGeoCameraCapabilities const&, QMap<QString, QVariant> const&)
```
I.e. `/usr/lib/x86_64-linux-gnu/qt5/plugins/geoservices/libqtgeoservices_googlemaps.so` and `/usr/lib/x86_64-linux-gnu/libQt5Location.so.5` have different `QGeoMapType::QGeoMapType()` signatures.

When I look where these come from:
```
$ dpkg -S /usr/lib/x86_64-linux-gnu/qt5/plugins/geoservices/libqtgeoservices_googlemaps.so
subsurface: /usr/lib/x86_64-linux-gnu/qt5/plugins/geoservices/libqtgeoservices_googlemaps.so
$ dpkg -S /usr/lib/x86_64-linux-gnu/libQt5Location.so.5
libqt5location5:amd64: /usr/lib/x86_64-linux-gnu/libQt5Location.so.5
```

--> the `subsurface` package (I use that to quickly test against the release) clashes with the `libqt5location5` package.

bstoeger, 2018-11-02:
Aaand... Indeed `sudo apt remove subsurface` fixes the problem!

Thank you @neolit123 and @mturkia!

closed 2018-11-02 by bstoeger