subsurface/subsurface #1849: OSTC sport dive log import fails

mkohlrusch, 2018-11-08:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When downloading divelogs from my OSTC sport It fails. The device shows 'Download mode enabled' and after a few seconds 'exited'. Subsurface show an error 'error importing divedata'.
The model and serial number are displayed correctly, but the firmware shows as 2605, should be v10.45?
Downloading divedata worked without any issues previously. 

### Operating system:

Besturingssysteem: macOS High Sierra (10.13)
Taal: nl-NL

### Subsurface version:

Subsurface 4.8.3.0, official release. Also happened on previously releases. It worked in the past, but do not know the version that last worked correctly.

### Steps to reproduce:

1. Import -> Import from divecomputer
2. On the device -> Menu -> Start Bluetooth
3. Press the 'download' button. All settings are correct

### Current behavior:

Subsurface gives an error when downloading divelogs from the computer

### Expected behavior:

Subsurface imports divelogs without any error message

### Additional information:

Log and dump added.
[subsurface.dump.log](https://github.com/Subsurface-divelog/subsurface/files/2562013/subsurface.dump.log)
[subsurface.log](https://github.com/Subsurface-divelog/subsurface/files/2562014/subsurface.log)

### Mentions:

mkohlrusch, 2019-01-29:
Downloaded the latest version today, but still the same error. Can I help to solve this issue?

atdotde, 2019-01-29:
Are you connecting via Bluetooth or Bluetooth LE (you can tell the later by the longish address)? Only the latter works for me currently on Mac.

atdotde, 2019-01-29:
This is classic BT. Let the discovery run a bit longer and you will find the OSTC on BTLE.

glance-, 2019-01-29:
Post the logs from a regular download to be able to see whats going on. Looks like the dump goes just fine, so I'm betting on that its broken data, or some parser error.

mkohlrusch, 2019-02-01:
I also uploaded a regular dump when I created the issue. Is that what you're referring to? If not, how can I create such a log.
With the respect to data/parser error. I've not been able to upload a dive since my computer was repaired (broken pressure sensor). The first dive is a 67.4m dive in 10 minutes. Could that be the cause?

janmulder, 2019-02-01:
That pressure chamber "dive" from HW should normally not matter. I agree with Anton (@glance-) that is seems like broken data, or a very unlikely parser error (as nothing has changed there for a long time).

That said. I use a very similar OSTC3, and when I produce a "libdivecomputer log" I see things that I do understand. For example, sending 0xBB byte (go into download mode), a response 0xBB from the OSTC3 and a 0x4D (end of response byte) on the start of communication.

The "libdivecomputer log" you uploaded before is made from the same download session as the "libdivecomputer dump log". Both these files look just fine.

I would be interested (and I think that is what Anton means as well) is a "libdivecomputer log" of a normal (and failing) download session, and do *not* select the dump option as well. In addition, start Subsurface from the command line and add `-vv` as parameters for very verbose logging, and post that output as well.

mkohlrusch, 2019-02-01:
I've attached a regular dumpfile with -vv. 
[subsurface-01022019.log](https://github.com/Subsurface-divelog/subsurface/files/2821272/subsurface-01022019.log)

janmulder, 2019-02-01:
Ok, analyzed this file. It starts just fine: 1) comms started, 2) Hardware + features are reported, 3) Version/Identity is reported 4) Then the compact headers are asked, and 4k data is returned  (as expected).

Now, the first warning: `WARNING: Unexpected empty header found. [in ../../src/hw_ostc3.c:751 (hw_ostc3_device_foreach)]` and things go downhill fast from here. Dive number 0x4C is requested, and that one is not reported any more.

Conclusion: Anton was right. Broken data, that we obviously cannot parse.

But the main question now is: where do we go from here? When there are no relevant dives on the OSTC, a reset logbook might be an easy option to try. But when there are relevant dives ... there might be a way to read (raw) data from the OSTC using libdivecomputer's dctool in order to salvage most of the relevant dives.

mkohlrusch, 2019-02-01:
I have about 30 dives on my computer. If there is a way to keep the raw data I could make a dump and then reset the dive logbook. I can then later try to import the dives from the raw data (with a little help?).
Or I will just enter the dives in Subsurface manually and reset the dive logbook.

glance-, 2019-02-01:
I've had bad data on my OSTC3 and I went over the top and "repaired" the data. Not that I recommend anyone doing that.

My suggestion is to hack up your own libdivecomputer which skips the bad dive data in a way that suites you.

Another way which might work, is to use dctool and download all the dives in "raw" format, and then selectively parse the ones that you can.

Libdivecomputer isn't that tolerant against bad data. It's a door that swings both ways. It makes it easier to find bugs and bad assumptions in the libdivecomputer code, but its a PITA if your dc contains bad data.

mkohlrusch, 2019-02-04:
Thank you all in helping me trying to solve this, I appreciate it a lot!

I'll start to record my dives manually in Subversion. I'm still thinking whether I will pick up the challenge (go over the top) and try to repair the data. So give me some time, I'll definitely update the issue with my findings.

mkohlrusch, 2019-02-04:
Just tried dctool, but with the same kind of error, alas. 

Opening the I/O stream (serial, /dev/tty.OSTCs12677-SPP).
Opening the device (Heinrichs Weikamp OSTC 3).
Registering the event handler.
Registering the cancellation handler.
Downloading the dives.
Event: progress 0.00% (0/4194304)
Event: model=18 (0x00000012), firmware=2610 (0x00000a32), serial=12677 (0x00003185)
Event: progress 0.02% (1024/4194304)
Event: progress 0.05% (2048/4194304)
Event: progress 0.07% (3072/4194304)
Event: progress 0.10% (4096/4194304)
WARNING: Unexpected empty header found. [in ../../source/src/hw_ostc3.c:751 (hw_ostc3_device_foreach)]
Event: progress 0.02% (4096/17043844)
Event: progress 0.02% (4097/17043844)
ERROR: Failed to receive the answer. [in ../../source/src/hw_ostc3.c:323 (hw_ostc3_transfer)]
ERROR: Failed to read the dive. [in ../../source/src/hw_ostc3.c:816 (hw_ostc3_device_foreach)]
ERROR: Error downloading the dives. [in ../../source/examples/dctool_download.c:228 (download)]
ERROR: Timeout

janmulder, 2019-02-05:
With dctool failing I think the only way to get things going again is, as Anton described above, hack in libdivecomputer to ignore the offending data., which obliviously requires to build things from source.

jefdriesen, 2019-02-22:
When I analyze the logbook entries, I see this:
```
B817001006110D178A1821001F010024 -> 1
370E001007180A03CA020D0018020024 -> 2
...
470C001205060B0863050B002A4C0024 -> 76
9FBFFB7FFFFF7FFFFFFF7FFFFF7F37BD -> 14207
E7090012051C1116541A0A00164E0024 -> 78
...
BC2900120A100B2D3E093E0008690024 -> 105
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF -> empty
...
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF -> empty
```
It's very obvious there is a problem with dive #77. The dive number is clearly wrong, but the rest of the data looks suspicious too. Now, because this entry has the highest dive number, it is considered to be the most recent dive. But when we try to download that dive, we don't receive any data. I suspect that's because the ostc is unable to locate the profile data (because the pointers in the logbook entry are probably bogus too).

You mention you have about 30 dives. I suspect those are dive #78 to #105 after the problematic entry.

What happens you try to view your dives in the internal logbook on your ostc? Can you still view those 28 dives?

This can probably be fixed by erasing the bogus logbook entry. Can you send me a full memory dump of your dive computer? I need that to verify the correct memory address of the bogus entry.

jefdriesen, 2019-02-22:
I just noticed you already attached a memory dump. The log extension was a bit misleading. Anyway to erase the bogus entry, proceed as follows:

First, backup the entry, just in case:
`dctool -vv -l read.log -f ostc3 read -a 0x24C000 -c 0x1000 -o backup.bin /dev/tty.OSTCs12677-SPP`
Then, generate the empty data file:
`dd if=/dev/zero bs=4096 count=1 | tr "\000" "\377" > empty.bin`
And finally, overwrite the entry:
`dctool -vv -l write.log -f ostc3 write -a 0x24C000 -c 0x1000 -i empty.bin /dev/tty.OSTCs12677-SPP `

mkohlrusch, 2019-02-22:
Hi Jef,
Thanks for looking into this. I have 28 dives in my OSTC, numbered from 1 to 28. Number 1 is my last dive, number 28 is the 'testdive' by Heinrichs Weikamp.
I can all the dives on my computer and if I select one I can see the profile and other information. Did not check all the dives, but the first and last ones, and a couple in between.

mkohlrusch, 2019-02-22:
O, i see you already commented, will try your solution now.

mkohlrusch, 2019-02-22:
Hi Jef,
Question, in step 1 a memory dump of 4k is generated. Step 2 (dd) generates a 8k empty.bin. I would expect a 4k block (bs=4096 count=1). If I leave out the '|' and the rest it creates a 4k file. Can you help me on this?
Kind regards, Martin

jefdriesen, 2019-02-22:
Just use the attached [empty.bin.gz](https://github.com/Subsurface-divelog/subsurface/files/2893948/empty.bin.gz) file. Don't forget to decompress the file first:

`gzip -d < empty.bin.gz > empty.bin`

mkohlrusch, 2019-02-22:
Yes! I was able to import my dives. Thank you all for your assistance, very much appreciated!

@Jef: I'll make a donation, you saved me a lot of time.
@Subsurface-guys: Don't see a donation option, is there any?
@All: If I can help you in the future, feel free to ask.

Kindest regards,

Martin

atdotde, 2019-02-22:
Martin,

> On 22. Feb 2019, at 14:01, mkohlrusch <notifications@github.com> wrote:
> 
> Don't see a donation option, is there any?

everything Subsurface is paid for by Dirk. But contributing your time would be very much appreciated! Besides the obvious addition of code, improving the user manual, answering questions in the support group or simply testing out new versions to make sure we don’t break stuff is very much welcome.

Best
Robert

janmulder, 2019-02-24:
Lets close this. All is solved with some magic raw writing to the device. Jef is a real master in binary patching up some corrupt data.

closed 2019-02-24 by janmulder