subsurface/subsurface #1852: Crash on FB login page on Mac

atdotde, 2018-11-11:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

I was using the official 4.8.3 binary (since my own builds don't have FB integration). Upon selecting "Share on -> Facebook", it loads the FB login page, shows the form and then, after about 2-3 sec crashes reproducibly. The stack dump is unspecific deep inside WebKit with the first Subsurface code 46 levels in being updateProgress + 512. Honestly, I have no clue what is going on.

### Operating system:

MacOS Mojave in MBP

### Subsurface version:

### Steps to reproduce:

### Current behavior:

### Expected behavior:

### Additional information:

### Mentions:

atdotde, 2018-11-16:
Similar crash when opening the manual. Furthermore, I can report this to happen both on my iMac (High Sierra) as well as MBP (Mojave). Sounds to me like a WebKit problem. BTW, 4.8.4. crashes as well. @dirkhh do you have a chance to look into this or do I have to bite the bullet and build WebKit for a recent Qt myself (unknown time frame)?

dirkhh, 2018-11-16:
It appears that this has been broken since at least 4.8.1 and no one has complained - which makes me wonder how many people would care if we just killed the stupid FB integration :-)

So yes, I need to go back and figure out what broke WebKit again on Mac. But seriously, of our 2000+ Mac users no one uses printing, FB, the manual? Strange. Very strange.
Also, clearly, we don't test our own software at all. But then, what else is new.

atdotde, 2018-11-17:
Yeah, looks good. Thanks Dirk!

closed 2018-11-17 by atdotde