subsurface/subsurface #1887: Editing location on mobile does not work

mturkia, 2018-12-16:
### Describe the issue:

- [x] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When editing a dive on subsurface-mobile, the location is not saved.

### Operating system:

subsurface-mobile on desktop, probably the native iOS and Android builds as well, but haven't tested

### Subsurface version:

master

### Steps to reproduce:

- Open divelist.
- Choose a dive and edit the location
- Save
- Check if the location was changed on divelist and dive details

### Current behavior:

The location sticks to the original one on both dive list and details

### Expected behavior:

Location is updated to the new value
### Additional information:

### Mentions:

bstoeger, 2018-12-28:
Grrrr.... I have a feeling that this has already been fixed numerous times. ☹

bstoeger, 2018-12-28:
Bug introduced in 68961a169efc37039cd3fda334efb9ad9927444f, will check it out later today.

closed 2018-12-30 by dirkhh