subsurface/subsurface #1892: Subsurface for Fedora 29 / Buildservice

neufeind, 2018-12-19:
While various Fedora-versions and also "rawhide" are available, the current release Fedora 29 is missing. Could you maybe activate it for the buildservice?

neufeind, 2018-12-20:
Thanks, works like a charme. Trying to install the Fedora 28-packages didn't work because of dependencies, but the F29-build works great.
Until now I used the AppImage-package on Linux, which also works great. (You say on the website you want feedback). But where available, as here, I prefer cleanly built versions using the system-libraries and installed from a repo.

Super, impressive, quick ... thanks a lot.

neufeind, 2018-12-20:
PS: Then maybe also mention it on the downloads-page or similar locations: https://subsurface-divelog.org/download/

closed 2018-12-20 by neufeind

dirkhh, 2018-12-20:
Great point. I wanted to wait for the build to have finished and be tested by someone. And in my time zone it's not even 8am :-)
Will do.