subsurface/subsurface #1896: Bluetooth is not enabled on Subsurface mobile

JMZalewski, 2018-12-22:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

I'm using suunto eon steel and a xiaomi mi6 (android 8.0.0 OPR1.170623.027 with MIUI Global 10.0.1.0).
I can connect the dive computer to my mi6 using movescount from suunto, but when I start subsurface mobile I can see a message "Bluetooth is not enabled" or "Bluetooth is not available"

### Operating system:

XIAOMI MI6 -> android 8.0.0 OPR1.170623.027
Suunto Eon Steel -> 1.6.5

### Subsurface version:

Subsurfae-mobile 2.1.5 (4.8.4.0)

### Steps to reproduce:

1. Start subsurface-mobile
2. Click on computer connect icon on the bottom or swipe the screen to the left
3. The error message is visible on the right side of "Rescan" button

### Current behavior:

On the connections list is only FTDI displayed

### Expected behavior:

### Additional information:

### Mentions:

dirkhh, 2018-12-22:
There are a very small number of Android devices on which the libraries that we are using to build Subsurface-mobile cannot seem to talk to the Bluetooth stack on the device. This is fundamentally a Qt problem and not something that Subsurface can solve.

JMZalewski, 2018-12-22:
You may have right, but:
In the function: BTDiscovery::BTDiscoveryReDiscover
in case of Android devices QBluetoothLocalDevice is used,
but in case of iOS  the usage of QBluetoothLocalDevice is skipped with following comment: 
	// for iOS we can't use the localBtDevice as iOS is BLE only
	// we need to find some other way to test if Bluetooth is enabled, though
	// for now just hard-code it
It's exactly my case. My device is Bluetooth low energy only! So it should be handled similar way.

There is already a workaround for iOS. All I need is too use the same for my android device. In my  opinion simply the check of QBluetoothLocalDevice could be omitted also in case of Android devices.

I think it should be designed in following way:
1. Use QBluetoothDeviceDiscoveryAgent without checking QBluetoothLocalDevice
2. If it doesn't work check the QBluetoothLocalDevice for the BT status (connectable, valid, ect.) and display a proper error message.

Such solution should work well on iOS, BLT Android and standard Android devices...

janmulder, 2018-12-23:
Are you telling us that you have an Android device, that claims to conform to Bluetooth 5, but at the same time is BLE only? I have a hard time believing that, as Bluetooth 5 simply supports Bluetooth Classic. Obviously, I can be broken on the firmware you are running on the device.

This said. Can you post the debug log (go to about screen, copy logs to clipboard button, and paste it here)?

Obviously, being an open source project, you are free to provide a fix for the issue to are seeing, as we cannot reproduce this.

JMZalewski, 2018-12-23:
The log file is attached.
[subsurface.log](https://github.com/Subsurface-divelog/subsurface/files/2705988/subsurface.log)

janmulder, 2018-12-23:
A new version can be downloaded from https://transfer.sh/681eU/subsurface-mobile-build-arm-debug.apk

This is a version up to the current master + 1 commit as a test to fix the problem. As it is a debug build, It cannot be installed over the version that came from Google's Play Store as it is unsigned. So first remove that one, install this debug one with adb, obviously you also need to allow installing apps from non trusted sources (ie.  other that the official store) when installing over adb.

As said before. We cannot reproduce the problem, meaning that this possible fix will might be abandoned when not tested by anybody that could reproduce the problem at first.

JMZalewski, 2018-12-23:
I've done following:
1. uninstall subsurface-mobile from GooglePlayStore
2. copy apk into my device
3. install the apk

When I tap the installed  app icon i recive an error message: "Fatal error, your application can't be started." The detailed error description you can find in attachment.
[image]
Should I install some additional dependecies to run the debug version?

janmulder, 2018-12-23:
I just verified the same Travis build apk, and it fails for me as well. Travis is the external CI service we use in the project. Apparently, that build is broken as you do not need to install extra things to run a debug build. Very strange (@dirkhh: do you understand this?). I'm one of the few guys that build his own Android version (for testing and development), so I can provide you with an apk that runs, but build by me. Give me 10 minutes, and I will post a link where to download my build.

janmulder, 2018-12-23:
Ok, the one build be me and just verified on by device: https://transfer.sh/HSO9L/subsurface-mobile-build-arm-debug.apk

dirkhh, 2018-12-23:
@janmulder I do test the builds from Travis every once in a while and I know that they worked "maybe a month" ago. 
As you know I am JUST in the process of moving to a different build structure for Travis, so let's focus on getting that new one to work...

JMZalewski, 2018-12-23:
I've just made a test and it works now!
@janmulder: it looks you were right about the bluetooth LE :)

Guys, thanks for fast reaction! Great job!

closed 2018-12-23 by JMZalewski

janmulder, 2018-12-23:
Nice. Thanks for testing. The fix will go into the next version. Notice that you need to remove the debug version again, when you want to install a new version from the Google Play store.