subsurface/subsurface #1910: Main Menu: Move Share on

seanwhe, 2018-12-31:
### Describe the issue:

- [ ] Bug
- [x] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

The Share on item in the Main Menu is a feature not a generic menu. Suggest relocation to the Log menu and/or inclusion in context menus within the Dive List or Dive Profile panels.

Move the setup of the Share on into the File>Preferences dialog under Social.

### Operating system:

### Subsurface version:

v4.8.4

### Steps to reproduce:

Not Applicable

### Current behavior:

Share on is a main menu

### Expected behavior:

- Move Share on to sub menu Log > Share
- Move Share on config/setup to File > Preferences dialog

### Additional information:

### Mentions:

dirkhh, 2018-12-31:
I think that's mostly a matter of taste.
To the best of my knowledge sharing to Facebook is currently broken and no one cares enough to fix it...

neolit123, 2019-01-02:
my vote goes to removing the "share on facebook" functionality from the code base along with the "share on" menu.

> Then again, is social sharing a core intent behind the software?

i don't think that social sharing has seen much demand or use, so i would say no.

neolit123, 2019-02-01:
the menu is removed for now because weare  removing the FB feature.
https://github.com/Subsurface-divelog/subsurface/pull/1958

closed 2019-02-01 by neolit123