subsurface/subsurface #1911: New Logbook: Append .xml file extension when saving a new logbook file

seanwhe, 2018-12-31:
### Describe the issue:

- [ ] Bug
- [x ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When saving a new logbook for the first time, Subsurface default is to create a XML file named by the user. The resulting file name does not have an extension.

### Operating system:

Not Applicable

### Subsurface version:

v4.8.4

### Steps to reproduce:

1. Create a New Logbook
2. Save the new logbook
3. Name the new logbook 'testing' without file extension
4. Click Save button
5. At the command prompt ~/.subsurface$ ls 
cloudstorage  googlemaps  hashes  printing_templates  testing  thumbnails

### Current behavior:

Saves the file without .xml

### Expected behavior:

Saves file with .xml
If .xml is supplied in naming strip the .xml from the filename to aviod testing.xml.xml

### Additional information:

### Mentions:

dirkhh, 2018-12-31:
actually, I think the expected behavior would be:
if suffix given, use whatever suffix is given.
if no suffix given, use `.ssrf`

seanwhe, 2019-01-01:
I did not know there is a suffix for Subsurface files.

Would "use whatever suffix" be a good idea going forward?

Either way I think it always best to remain consistent. Use either .xml or .ssrf but keep it consistent.

neolit123, 2019-01-02:
> When saving a new logbook for the first time, Subsurface default is to create a XML file named by the user. The resulting file name does not have an extension.

> Operating system:
Not Applicable

what OS is this? is this a bug?

on Windows saving a new log without writing the extension in the file box, defaults to ssrf as expected.

seanwhe, 2019-01-02:
Could be. Sorry I don't have a Windows.

$ uname -a
Linux sean-home 4.15.0-43-generic #46-Ubuntu SMP Thu Dec 6 14:45:28 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

bstoeger, 2019-05-11:
I cannot reproduce this on Linux.
On "save as", the extension of the current file is taken.
On "new logbook" followed by "save as" if I omit the extension, a ".ssrf" is added.

mikeller, 2026-10-17:
Hi @seanwhe,

We tried to reproduce this and couldn't: on "New Logbook" followed by "Save As" with no extension typed, Subsurface already adds ".ssrf" correctly, confirmed on both Windows and Linux.

Closing this as not reproducible / already correct. Please reopen with your exact OS, build, and the save dialog you used if you still see a file saved with no extension.

closed 2026-10-17 by mikeller