subsurface/subsurface #1918: Check for updates: server side broken

seanwhe, 2019-01-02:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

When performing a "Check for updates" the following error is returned.
Subsurface was unable to check for updates.
The following error occurred:
Error transferring http://subsurface-divelog.org/updatecheck.html?os=linux&version=4.8.4.124&uuid=999a51fc-691e-4f97-98f3-13bfbe8602a3 - server replied: Proxy Error

### Operating system:

### Subsurface version:

v4.8.4

### Steps to reproduce:

Help > Check for updates

### Current behavior:

The following error occurred:
Error transferring http://subsurface-divelog.org/updatecheck.html?os=linux&version=4.8.4.124&uuid=999a51fc-691e-4f97-98f3-13bfbe8602a3 - server replied: Proxy Error

### Expected behavior:

Return information as to current version and whether there is a new version available

### Additional information:

### Mentions:

neolit123, 2019-01-02:
works for me. are you behind a proxy by any chance?
preferences -> network has a proxy setup.

dirkhh, 2019-01-02:
Nope, my backend processing was crashing and I fixed it a few minutes ago.
The bug report was correct, I just already addressed the issue

closed 2019-01-02 by dirkhh