subsurface/subsurface #1922: Preferences > General > Video thumbnails

seanwhe, 2019-01-02:
### Describe the issue:

- [x ] Bug
- [ ] Change request
- [ ] New feature request
- [ ] Discussion request

### Issue long description:

Installed on a system that does not yet have FFMPEG installed, the "Extract video thumb nails" checked by default.

### Operating system:

Linux 4.15.0-43-generic #46-Ubuntu SMP Thu Dec 6 14:45:28 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

### Subsurface version:

v4.8.4

### Steps to reproduce:

1. Install on a system that does have FFMPEG installed
2. File > Preferences > General

### Current behavior:

Default checked when FFMPEG is not found

### Expected behavior:

Default unchecked when FFMPEG is not found

### Additional information:

### Mentions:

seanwhe, 2019-01-28:
I think it should be unchecked and should warn that FFMPEG is found or not.

If the user tries to check it, it should attempt to detect ffmpeg and if fail give some message to say, install it.

IMHO cosmetics count in a big way to usability.

bstoeger, 2019-01-30:
We probably shouldn't nag users that will never add a video (the vast majority?) with `ffmpeg` related messages. We could show a message the first time they attempt to add videos and `ffmpeg` doesn't work though. Likewise if they actively try to turn on the option.

@dirkhh: What do you think?

dirkhh, 2019-01-31:
I would be annoyed if Subsurface babbled about ffmpeg when I have no interest to ever attach a video to a dive. So no, this should happen the first time you use it.

bstoeger, 2019-08-03:
We now warn if the user attempts to set a non-working ffmpeg and give a better error message if the user tries to add videos without having ffmpeg installed.

It's not exactly what the reported wanted, but I think it's OK from an UI point of view?

mikeller, 2026-10-17:
Hi @seanwhe,

Rather than changing the checkbox's default, we went with a different fix: Subsurface now warns if you try to set a non-working ffmpeg path, and gives a clearer error message if you try to add videos without ffmpeg installed, so you're told what's wrong at the point you actually try to use the feature, rather than the checkbox silently defaulting to on.

Closing this as addressed via that approach. Please reopen if you're still not getting a clear warning when trying to add a video without a working ffmpeg installed.

closed 2026-10-17 by mikeller